openrocket/openrocket #999: Unit rounding creating discontinuity issues.

hcraigmiller, 2021-08-24:
See, https://github.com/openrocket/openrocket/issues/871#issuecomment-899110349

SiboVG, 2021-09-27:
#1036 is an issue that also arose from the confusion of rounding off the input fields to single decimal numbers.

SiboVG, 2022-01-31:
Just want to put out a random thought before I forget it: maybe it's an idea to be able to right-click a value entry box and set the number of displayed decimals?

Demo:
[image]

This could help users who want to display higher precision numbers than the default ones (like in #1036).

neilweinstock, 2022-10-18:
Ugly.

But also reinforces (for the umpteenth time) that _every single warning and error_ should specify exactly where it comes from.  Just saying "Discontinuity error" without identifying the two components and the offending values is super dumb.

artao5, 2022-10-18:
> Additional information about this bug is derived from a post on the FaceBook OpenRocket group. The user created a design that exibited a discontinuity error with no visible explanation as to why; all outer diameters appeared to be the same.
> 
> Deconstruction (stripped file) revealed that the discontinuity related to a disparity between the outer diameter of the Payload Bay (an Estes component) and the Sustainer Airframe (a custom component). Changing the characteristics of the Sustainer Airframe, (1) checking the automatic box removes the discontinuity warning, (2) re-entering the outer diameter does NOT remover the warning, (3) manually changing the outer diameter (entering the change), then re-entering the original outer diamter removes the discontinuity warning. Looking at the saved file with the discontinuity warning, the Payload Bay (Estes tube) has a radius of 0.0093472 and the Sustainer Airframe (custom component) has a radius of 0.00935.
> 
> [Stegeman, Vaughn.Canopus Archangel Sounding Rocket A03.ork.txt](https://github.com/openrocket/openrocket/files/9814216/Stegeman.Vaughn.Canopus.Archangel.Sounding.Rocket.A03.ork.txt) (Original .ork file)
> 
> [Stegeman, Vaughn.Canopus Archangel Sounding Rocket A03.Stripped.ork.txt](https://github.com/openrocket/openrocket/files/9814175/Stegeman.Vaughn.Canopus.Archangel.Sounding.Rocket.A03.Stripped.ork.txt) (Stripped .ork file)

Thanks for posting this. I am the designer of the Canopus Archangel. :)

hcraigmiller, 2022-12-16:
In reviewing numerous designs posted on several threads, I found the most common continuity error could not be seen by looking at the diameter values. There are likely two possible reasons for this, (1) a portion of the airframe was scaled and then a standard airframe component added (for discontinuity purposes, the scaled diameter is not rounded up or down to the number of decimal points in the displayed value) and (2) manually entered values are beyond the precision of the decimal points in the displayed value, and are slightly different (using inches for instance, 1.1875 and 1.1876 both display as 1.188, triggering a discontinuity error).

The fix would seem to be for the discontinuity flag to trigger the warning when the precision of the value displayed (applied to the component diameters, as if rounding the component diameters to that level of precision) is not the same; the actual, more precise value would be retained for all other purposes. Using the example above, for a three decimal point precision, because both 1.1875 and 1.1876 round to 1.188 no discontinuity error would be displayed.

neilweinstock, 2022-12-16:
I'm not convinced that the 4th decimal place (5th significant digit in this case) is worth preserving at all... but yes, at the very list, limiting the discontinuity checking to the number of displayed decimal places would be a start to eliminating the absolute worst of the warnings, where the visible dimensions are identical.  I don't know off the top of my head if there are any particular pitfalls in doing this.

JoePfeiffer, 2022-12-16:
I'm not sure how big a discontinuity is worth telling the user about -- but if it's worth displaying a warning, it's worth displaying as different in the text fields.  My intuition is that a difference in the fifth significant digit doesn't matter (my caliper doesn't display that many digits, and I sure wouldn't be able to measure accurately at that level if it could), but the one thing I'm sure of is that if we display the diameters as equal, the software should regard them as equal and not put up the warning.

neilweinstock, 2022-12-16:
>  the one thing I'm sure of is that if we display the diameters as equal, the software should regard them as equal and not put up the warning.

It's impossible to argue with that.  The body discontinuity warning could use more work in the future but implementing this one common-sense behavior would seem to be a good small step for this release, if it's not too much of a mess to implement.

PR #1931 (2023-01-05): Add open airframe warning and use text compare to find airframe discontinuities
Adds a warning if the forend of an airframe is open (ie there isn't a nose cone or equivalent at the start of an airframe or pod)

Changes the test for diameter discontinuity to simply be whether the two diameters end up with the same display string.  This will cause some surprises if a user goes back and forth between two units, but that seems less problematic than the current situation where a discontinuity error can be set when the two diameters are displayed as being equal.

Finally, it turned out that the code [... cut]
closed 2023-01-05 by JoePfeiffer