openrocket/openrocket #1025: New feature request: Add 'delete' hotkey for deleting component

SiboVG, 2021-09-16:
I often find myself wanting to delete components by first selecting it and then hitting the 'delete' key on my keyboard. Right now, this doesn't do anything to the component, but it would be nice to have this key command implemented, since it is quite a natural response for wanting to delete something in a program.

neilweinstock, 2022-01-26:
I just tried this on latest unstable and it worked.  Selected a component, hit <delete>, and it disappeared.

Was this fixed at some point?

hcraigmiller, 2022-01-26:
Speaking in Windows terminology, Ctrl C (copy), Ctrl V (paste), Ctrl X (cut) have worked for quite some time, although not without some normal limitations, as I recall. For example, using _A simple model rocket_, a component (or component tree) attached to the body tube can be copied, copied and pasted, cut, or cut and pasted anywhere on that body tube. The normal limitation is that you cannot paste a component onto another incompatible component, such as pasting trapezoidal fins onto a nose cone (you can paste freeform fins onto a nose cone).  I don't recall any misbehavior in functionality, and didn't find any today.

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

SiboVG, 2022-01-26:
> I just tried this on latest unstable and it worked. Selected a component, hit , and it disappeared.
> 
> Was this fixed at some point?

Hm I see, I used the backspace button instead of delete (Fn + Backspace on Mac). With the Fn + Backspace, deleting a component works

neilweinstock, 2022-02-01:
Should we close this one?

hcraigmiller, 2022-02-01:
It works in Windows. If it works in Mac then I think we're good.

SiboVG, 2022-02-01:
Yep, this can be closed, sorry for the initial confusion; Mac and their crazy delete shortcut 🙃

closed 2022-02-01 by SiboVG