openrocket/openrocket #1036: Question: How to increase lat/lon resolution in the initial simulation parameters?

Sparrowhawk120, 2021-09-26:
I'm experimenting with determining the landing location of my high power rockets, and the single decimal precision provided puts my initial starting point almost exactly 2.00 miles away from where I need it to be, which is an obvious issue. I couldn't find anything obvious in the code about where the lat/lon values are being truncated because they are stored as doubles which should be able to hold quite precise coordinates. Can anyone help me?

SiboVG, 2021-09-27:
Is it correct that your situation is that e.g. for the latitude, you want to enter '45.11111 °', but the input field "rounds" things off to '45.1 °'?
[image]
 
If so, then this issue is related to [this comment](https://github.com/openrocket/openrocket/issues/871#issuecomment-899110349). The thing is, OpenRocket only displays 45.1, but if you type in 45.11111, then internally, OpenRocket will use 45.11111 instead of the 45.1 that is displayed in the GUI. So you can actually have all the precision that you want by just typing in the value that you want, but it will only be displayed with a single decimal precision in the GUI.

You can see this in action in the image below. I typed in '45.12345' as latitude value and after exporting the simulation data, you can see that OpenRocket does use the precise value that I entered.
[image]

I propose we close this issue, as a similar issue has already been created (#999). I will link this issue to #999.

Sparrowhawk120, 2021-09-27:
Thank you for your help! I was looking to use extremely precise coordinates as it is a small field so 3 decimal places is actually not enough, am I able to input, for instance, [38.829970211521044, -77.8074957811569] (the exact location of my launch site) ?

Sparrowhawk120, 2021-09-27:
^to add on, when I enter the above values they truncate to 38.230, -77.807

SiboVG, 2021-09-27:
Well, once again, internally OpenRocket does use your precise coordinates. I printed the variables just before they got converted to a CSV. The result can be seen below:
`38.82997022458307    -77.80911848464007`

It's not an exact match between the value that was entered, but I believe this has to do with rounding off errors caused by converting the input to SI-units and then converting it back for the export. The reason why the values are truncated is because the CSV-exported just by default takes 3 decimals as a maximum. I will create a new issue report for the ability to increase the number of decimals on export.

Sparrowhawk120, 2021-09-27:
Great, thank you so much!

closed 2021-09-27 by Sparrowhawk120