openrocket/openrocket #1037: New Feature Request: Ability for custom number of decimals on exporting simulations

SiboVG, 2021-09-27:
This issue is related to [comments in #1036](https://github.com/openrocket/openrocket/issues/1036#issuecomment-927984493).

Right now, when exporting the simulation data to CSV (go in the simulations tab, select a simulation, click the 'Plot / export'-button and in the 'Export data'-tab exporting the simulation), the variables have a fixed number of decimals (in the case of the comments to #1036, a maximum of 3 digits). Sometimes it's necessary for users to have more precise values reported back to them. Because OR internally uses more precise values than those that are eventually exported, it is merely a matter of implementing a **new feature for users to be able to control the maximum number of decimals used upon export**.

This feature could be implemented as a JSpinner in the 'Export data'-tab in which the user can set the desired number of decimals.

Sparrowhawk120, 2022-06-12:
Solved in #1427 , can we close this?

SiboVG, 2022-06-12:
> Solved in #1427 , can we close this?

Yes

closed 2022-06-12 by SiboVG