openrocket/openrocket #1039: OpenRocket Crashes.

AlbyVA, 2021-09-28:
There are various triggers that cause crashes, but this one always works like a charm.

1) Select View Type as 3D Figure
2) Select View Type as Side View.
3) App Crashes on my MacBook Pro.

Process:               OpenRocket [74426]
Path:                  /Applications/OpenRocket.app/Contents/MacOS/OpenRocket
Identifier:            net.sf.openrocket.startup
Version:               1.0 (1.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenRocket [74426]
User ID:               501

Date/Time:             2021-09-28 09:40:28.061 -0400
OS Version:            macOS 11.6 (20G165)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        DFB1EC6B-79FF-1806-EF1D-9908CC74B4D5

Sleep/Wake UUID:       8BFDC78E-5A04-47F6-ADB1-3451C07EFF5C

Time Awake Since Boot: 150000 seconds
Time Since Wake:       1400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000023642087b7d8
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x23642087b7d8:
    VM_ALLOCATE                 7c0480000-800000000    [  1.0G] ---/rwx SM=NUL  
--> 
    MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Application Specific Information:
abort() called

Thread 0 Crashed:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff203f492e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff204235bd pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff20378406 abort + 125
3   libjvm.dylib                  	0x0000000111354aeb os::abort(bool) + 25
4   libjvm.dylib                  	0x000000011147ae3e VMError::report_and_die() + 2304
5   libjvm.dylib                  	0x0000000111356716 JVM_handle_bsd_signal + 1131
6   libjvm.dylib                  	0x000000011135297b signalHandler(int, __siginfo*, void*) + 47
7   libsystem_platform.dylib      	0x00007fff20468d7d _sigtramp + 29
8   ???                           	0x00007ffee183cc98 0 + 140732681931928
9   ???                           	0x0000000111e809d4 0 + 4595386836
10  ???                           	0x0000000111e712bd 0 + 4595323581
11  ???                           	0x0000000111e712bd 0 + 4595323581
12  ???                           	0x0000000111e71302 0 + 4595323650
13  ???                           	0x0000000111e697a7 0 + 4595292071
14  libjvm.dylib                  	0x00000001111b8f06 JavaCalls::call_helper(JavaValue*, methodHandle*, JavaCallArguments*, Thread*) + 1710
15  libjvm.dylib                  	0x00000001111f0111 jni_invoke_nonstatic(JNIEnv_*, JavaValue*, _jobject*, JNICallType, _jmethodID*, JNI_ArgumentPusher*, Thread*) + 748
16  libjvm.dylib                  	0x00000001111e28df jni_CallVoidMethod + 363
17  libnativewindow_macosx.jnilib 	0x00000001648967ae 0x164891000 + 22446
18  com.apple.Foundation          	0x00007fff212cab81 __NSThreadPerformPerform + 204
19  com.apple.CoreFoundation      	0x00007fff2051b2bc __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
20  com.apple.CoreFoundation      	0x00007fff2051b224 __CFRunLoopDoSource0 + 180
21  com.apple.CoreFoundation      	0x00007fff2051b006 __CFRunLoopDoSources0 + 340
22  com.apple.CoreFoundation      	0x00007fff205199cc __CFRunLoopRun + 893
23  com.apple.CoreFoundation      	0x00007fff20518f8c CFRunLoopRunSpecific + 563
24  com.apple.HIToolbox           	0x00007fff287611f3 RunCurrentEventLoopInMode + 292
25  com.apple.HIToolbox           	0x00007fff28760f55 ReceiveNextEventCommon + 587
26  com.apple.HIToolbox           	0x00007fff28760cf3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
[... 973 lines cut ...]
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       508.0M       29 
__OBJC_RO                         70.3M        1 
__OBJC_RW                         2496K        2 
__TEXT                           270.5M      367 
__UNICODE                          588K        1 
mapped file                        1.0G      286 
shared memory                      772K       17 
===========                     =======  ======= 
TOTAL                              8.2G     2333 
TOTAL, minus reserved VM space     7.7G     2333 

Model: MacBookPro14,3, BootROM 429.140.8.0.0, 4 processors, Quad-Core Intel Core i7, 2.9 GHz, 16 GB, SMC 2.45f5
Graphics: kHW_IntelHDGraphics630Item, Intel HD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro560Item, Radeon Pro 560, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E434C47414C41522D4E5644
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E434C47414C41522D4E5644
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x173), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.8)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN 2, Ethernet, en9
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.4

SiboVG, 2021-09-28:
Was able to reproduce the crash on MacBook Air late 2015, Mac OS Big Sur 11.6, using the OpenRocket app (not the jar file, but the program from the [dmg-installer](https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.dmg))

SiboVG, 2021-10-13:
@neilweinstock just made a packaged version of the latest unstable release (thanks!). [Here is the link it](https://www.dropbox.com/s/gbd4tkk13l96kd5/OpenRocket%2020.x%20test.app.zip?dl=0). You should download the whole zip-file. When opening the application, you may get an error message saying that the app is damaged and can't be opened. This can be resolved by running the following command in your Terminal application: `xattr -r -d com.apple.quarantine <enter here the location to your .app file; don't include the greater than or less than symbols>`.

Tested the app and it didn't crash upon going to 3D view and back to side view. @AlbyVA if you could check if this is the case with you as well and whether the latest app solves any of the other issues that you had.

closed 2021-10-14 by neilweinstock