openrocket/openrocket #1044: calculateOgiveNoseInterpolator documentation mismatch?

Mudrost, 2021-10-07:
According to the documentation, in order to calculate the nose pressure drag OpenRocket performs a polynomial interpolation for the [1.0, 1.3] Mach range, then finds the constants "a" and "b" for the subsonic region. From what I can see, ```PolyInterpolator``` takes a set of y values and then their derivatives. In this case, instead of passing the CD value for M = 1 and then passing the derivative for M = 1, the code seems to be using the derivative for M = 1.3. 

[image]
Here, the term in red should be equal to the CD at M = 1 (sine of epsilon)

```// In the range M = 1 ... 1.3 use polynomial approximation
		double cdMach1 = 2.1 * pow2(sinphi) + 0.6019 * sinphi;
		
		double[] poly = conicalPolyInterpolator.interpolator(
				1.0 * sinphi, cdMach1,
				4 / (GAMMA + 1) * (1 - 0.5 * cdMach1), -1.1341 * sinphi
				);
```
However, in here, the CD at mach 1 (cdMach1) is actually the CD at mach 1.3, at least according to the documentation. Therefore, the first value being passed as y to the interpolator is correct (1.0 * sinphi), but not its derivative (second parameter). Am I missing something?

SiboVG, 2021-10-11:
> Great! I appreciate your time, @SiboVG<https://github.com/SiboVG>. Will keep an eye here for any updates. Oh, by the way, is that book a reference from the master thesis/documentation? Looks very interesting!

Yep, that's the book that was used in the thesis as the source of the Cd formulas.

SiboVG, 2021-10-11:
May I ask how you even found this potential issue? Seems like a very deep found.

JoePfeiffer, 2021-11-04:
I'm inclined to agree.  Unfortunately, the tech docs seem to take steps that are a little big for me to be comfortable that I'm following, so my thoughts are:
The docs don't explain where equation B.6 (Cd,M=1 = sin(epsilon)) comes from, and I don't see an explicit statement in Hoerner saying that (if I'm missing it, please point it out to me).  However, Hoerner's Figure 24 plotting Cd against sin(epsilon) at M=1 appears to be close to that.
As @Mudrost says, the 0.6019 found in the calculation of cdMach1 looks like the corresponding term in Hoerner's eq. 29 evaluated at M=1.3, so the "cdMach1" variable is really Cd at Mach 1.3.  When it's being passed in to conicalPolyInterpolator as the second argument it's correct but misleading; when it's being used in the calculation of the third parameter it's wrong.
@SiboVG I've got a hunch you've got a PR ready to go on this one; go ahead and submit it.  If not, I can.

SiboVG, 2021-11-04:
Thanks for the analysis @JoePfeiffer. Had the same conclusion about Cd,M=1, so Cd,M=1 = sin(epsilon) should be correct.

I will open up a PR for it ASAP.

PR #1057 (2021-11-04): [fixes #1044] Correct cdMach1 for ogive nose interpolator
This PR fixes #1044, where in the calculateOgiveNoseInterpolator-method, cdMach1 was implemented as being the cd at Mach = 1.3, instead of the intended cd at Mach = 1.
commit 6fb28818 (2021-11-04): [fixes #1044] Correct cdMach1 for ogive nose interpolator
The previous cdMach1-value was actually the cd at Mach = 1.3.
commit b4916bc2 (2021-11-04): Merge pull request #1057 from SiboVG/issue-1044

[fixes #1044] Correct cdMach1 for ogive nose interpolator
closed 2021-11-04 by JoePfeiffer