openrocket/openrocket #1054: Bug: Components cannot be selected in design window when in 3D modes

hcraigmiller, 2021-10-30:
When in a 3D view, a component cannot be selected by double-clicking that component. The component is not selected in either the component tree or design window, and the component configuration window does not open.

[Windows 10 Pro]

hcraigmiller, 2021-11-22:
**In the _Side View_:**

A single-click on a component in the tree highlights the component in the tree and 2D window, and a double-click on the same component opens the component configuration window. A single-click on a component in the 2D window highlights the component in the 2D window and tree, and a double-click on the same component opens the component configuration window for that component.

**In the _3D View_:**

A single-click on a component in the tree highlights the component in the tree and 3D window, and a double-click on the same component opens the component configuration window. However, a single-click on a component in the 3D window DOES NOT highlight the component in the 3D window or tree, and a double-click on the same component DOES NOT open the component configuration window for that component.

**Other actions observed in the _3D View_:**

A single-click on a component in the component tree selects that component in both the tree and 3D window, followed by a single-click in the 3D window, on any component or off (not on any component), deselects the component in both the tree and 3D view.

A single-click in the 3D view, on or off a component, causes what looks like a 3D window rewrite.

neilweinstock, 2021-11-22:
So this is not the same as #966, more like a Windows version of #754?

hcraigmiller, 2021-11-22:
First things first. What I describes as a possible 3D window rewrite, only affects components with a **_decal_** appearance; painted parts do not flash.

As to Issue #966, **_the behavior that I described in Issue #966 no longer exists in the Windows environment_**. Instead, the behavior described above occurs.

As to Issue #754, it would appear that both Mac and Windows now exibit the same bad behavior, Mac in Issue #754 and Windows in Issue #1054. Is there a way to combine these issues, or should the Windows comments be copied to Issue #754 and Issue #1054 closed?

PR #1471 (2022-06-22): [fixes #1054] Better register of double-click in 3D view
This PR fixes #1054 by implementing a custom double-click listener, where the wait time between two clicks is adjustable. 

Tested by @hcraigmiller.
commit 9350dbe1 (2022-06-22): Merge pull request #1471 from SiboVG/issue-1054

[fixes #1054] Better register of double-click in 3D view
closed 2022-06-22 by SiboVG