openrocket/openrocket #1055: Weirdness with saved simulation results upon opening a file

neilweinstock, 2021-11-03:
Sorry I couldn't think of a more precise title for this one.  Testing done with Unstable build from 27-Oct-2021.

Consider the following file.
[alcubierre_cranberry.ork.zip](https://github.com/openrocket/openrocket/files/7464841/alcubierre_cranberry.ork.zip)
When I open it up and then switch to the "simulations" tab, all simulations run except the last one:
[image]
This looks a lot like #915 except I've only been able to reproduce it in this one instance.  I'm not sure why any of the simulations re-run automatically in this case, perhaps something is wrong with the saved results in the ORK file.  But it definitely shouldn't do what it's doing, which is run all the sims except one.

Now consider this one.
[Firetail_update.ork.zip](https://github.com/openrocket/openrocket/files/7464847/Firetail_update.ork.zip)
When you open it up and switch to the simulations tab, all three show yellow (saved results):
[image]
None of them auto-run.

Now notice that in the rocket display, the current sim results all show N/A, despite the fact that the A3-4 is selected, and that motor *does* have a saved sim result.  In fact, if I change the motor selection, the display results stay at N/A.
[image]

I continue to be confused by the behavior of saved simulation results, and don't really have a clear idea of how they *should* work.  What's seen here is definitely wrong, though.

Have fun @SiboVG. :)

JoePfeiffer, 2021-11-03:
I've never fully understood the circumstances under which simulations are rerun...  I've gotten to where I just always rerun myself to be sure.  This is something that ought to be figured out, but it's not any sort of showstopper for next release.

neilweinstock, 2021-11-03:
In a previous PR, @SiboVG did a lot of work ensuring that the simulations get run at the right time.  Basically, the sims are now always kept up to date, something that 15.03 didn't manage very well.  These are a couple of remaining edge cases where they are all *not* kept up to date.

The saving of simulation results, on the other hand, is baffling to me.  Upon re-opening a design, the saved sim results are presented with yellow dots, but then *not* rerun unless you specifically request it.  It seems that inside the app, saved results live in some sort of never-never land in between "up-to-date, calculated results" and "stale results that must be updated".  IMHO this behavior is pointlessly confusing.  I presume it's done that way to avoid a time-consuming re-run of the sims every time you re-open the file, but I don't think that is sufficient justification for this weird behavior.

SiboVG, 2021-11-04:
Thanks for the report @neilweinstock! I'll see what I can do

SiboVG, 2021-11-05:
Looking inside the .ork-file of both designs, you can see that alcubierre_cranberry's simulations are marked as outdated, whilst firetail_update's are uptodate. This explains why alcubierre_cranberry's runs the simulations, whereas firetail_update doesn't.

[image]

However, Simulation 6 of alcubierre_cranberry is marked as uptodate. That's why this sim doesn't rerun. So the mechanism of only rerunning outdated simulations works; it's just the information saved in the .ork-file that is misleading.
[image]

If you manually re-run e.g. Simulation 1 and 3 of alcubierre_cranberry, save it and then open that file, you see that the simulations now saved correctly and do not have the outdated tag.
[image]

But, there are indeed some flaws in the system. I see one issue and one improvement in the behavior spotted by Neil:

1. (the issue) A simulation loaded from the .ork-file shouldn't produce N/A-values, because it has simulation data present
2. (the improvement) From a user's perspective, I wouldn't care if a simulation is loaded from the .ork-file or if it is just up-to-date. Moreover, it confuses users, as I was also confused when I first used OR and saw the yellow simulation status (yellow often denoting some sort of warning). I would just want green or red; simulation up-to-date or outdated.

So what I propose to do is:

- Fix the N/A issue
- Replace the 'loaded from file' simulation status to the 'up-to-date'-status (simplicity is key)

neilweinstock, 2021-11-05:
That sounds like a good plan to me, except for one problem: when saving a file, you have the option of not saving everything, or of using a large time increment:
[image]

Once again, this adds confusion.  If I save only primary figures, and then reopen the file, should the simulations really be considered up-to-date?  I don't know, maybe they should, but I'm not sure.

(also just now when opening up the alcubierre_cranberry file it gave an exception when running simulations, that's a new one ¯\_(ツ)_/¯)

SiboVG, 2021-11-05:
> (also just now when opening up the alcubierre_cranberry file it gave an exception when running simulations, that's a new one ¯_(ツ)_/¯)
[image]

SiboVG, 2021-11-06:
> Once again, this adds confusion. If I save only primary figures, and then reopen the file, should the simulations really be considered up-to-date? I don't know, maybe they should, but I'm not sure.

I think the saving simulation dialog stay; it can be useful to reduce file size by not saving the simulation data. But if you don't save the simulation, the simulations should be marked as outdated, because not all the simulation data is present to e.g. plot the simulation. In the latest unstable however, if you have updated all your sims and then save the file with only saving primary figures, no simulation data, the sims still have the 'Data loaded from file'-status. Since there isn't any real simulation-data in the .ork-file, you get N/A-values on the bottom-left corner of the rocket view.

So I will update the simulation-behavior to only have the up-to-date or outdated simulation status and when you don't save simulation data, the sims will get the outdated simulation status.

commit 6a3fa982 (2022-02-09): Merge pull request #1106 from SiboVG/issue-1055

[fixes #1055] Improve OR's simulation behavior
commit 7a778bf5 (2022-02-09): [fixes #1055] Save 'only figures' with status 'NOT_SIMULATED'
Better suited than the previous 'OUTDATED'
commit b52bcfa0 (2022-02-09): [fixes #1055] Change 'LOADED' sim status look to 'UPTODATE' look
commit cae55551 (2022-02-09): [fixes #1055] Fix simulation status after save only primary figures
Previously, if you saved a file with 'Simulated data to store: Only primary figures', the simulation status in the .ork file was still 'uptodate',but there was no sim data
closed 2022-02-09 by JoePfeiffer