openrocket/openrocket #1056: Feature Request: Forward and aft internal body tube extensions

neilweinstock, 2021-11-03:
In almost any design with a paper transition shroud, a body tube extends "inside" the rocket (e.g., inside the extension).  In OR right now, you need to model this by having the body tube end at the beginning of the shroud, and then adding an internal tube to represent the continuation of the body tube.
[image]

This means that the model diverges from reality in an important way.  A single component (body tube) is now represented in the model by two separate components.  Apart from the inconvenience of having to enter two separate parts in this way, you now cannot get a proper parts list from the program.  In my designs, I've often had to _very carefully_ add up the lengths of the different tube components to figure out how long to cut my tubes.  The addition isn't so hard, but you need to be careful to select the correct internal component to collect all the appropriate lengths.  This is the sort of thing the program ought to be able to do for you.

And so I propose that body tube components add two additional parameters: "forward internal extension" and "aft internal extension" that represent additional length of body tube that extend into the interior of the rocket at the forward and aft ends.  Obviously default would be zero, and in most cases the user will not need to touch them.  But they're there if you need them.  E.g.:
[image]

For simulation purposes, this is equivalent to adding an internal tube in front of or behind the body tube, with the same diameters.

It would be a nice quality-of-life improvement.

JoePfeiffer, 2021-11-03:
Something we should think about (AFTER next release) is whether we want the user to need to distinguish between inner and outer body tubes.  It ought to be possible for the aerodynamics calculator to recognize whether a tube is exposed to airflow.

neilweinstock, 2021-11-03:
That's a somewhat more radical proposal, because it would remove the base assumption that there is a set of components that make up the airframe, and are positioned linearly, in order.  If any body tube can be anything, then there would need to be a lot more positioning parameters; I'm not sure it would ultimately be an improvement.  Definitely would require some thought as to how to implement that.

My proposal above is much more modest, just a way to keep the model consistent with reality, and save effort when entering designs.

JoePfeiffer, 2021-11-03:
I'm thinking it would turn out to be less radical than it seems -- we already have more complicated positioning for just about everything else, including inner tubes.  Determining what part of a tube is an inner tube and what part is outer is going to be a little harder, but I think still doable.

neilweinstock, 2021-11-03:
Maybe. Seems like a bigger UX challenge than you are giving it credit for, but we’ll see. My feature request stands. 😀