openrocket/openrocket #1061: Bug# Edit window has dissapeared

JoshA314, 2021-11-20:
I'm running into an issue where I can no longer see the edit window. I used to use my laptop on a dual monitor setup with the second monitor extending my screen. I don't have it connected to the second monitor anymore and am running into an issue. Whenever I try to open the edit window, OpenRocket tries to open it on the second monitor that no longer exists and I can't see the edit window on my laptop screen. Even if I restart the program this issue still exists. Whenever I plug my laptop into an additional screen again the edit window reappears on the second window and I can use it again.

SiboVG, 2021-11-22:
Hi, thanks for the bug report. Did you use OpenRocket version 15.03 (the official release) or the latest unstable build to recreate this issue?

neilweinstock, 2021-11-22:
Try changing the screen resolution, then changing it back.

neilweinstock, 2022-01-26:
@JoshA314 do you have any update on this?

hcraigmiller, 2022-02-10:
No response from reporting party since 11/21/2021, recommend closure.

closed 2022-02-10 by SiboVG