openrocket/openrocket #1062: Clusters with asymmetric ignition do not simulate asymmetric thrust consistently

CarVac, 2021-11-27:
Version: OpenRocket 15.03 AppImage

I was curious about the effect of increasing the moment of inertia of a clustered rocket so I quickly drew up a rocket with a LOC 4" nosecone, a Blue Tube 4"x48" airframe, two Blue Tube 29mmx12" tubes as motor mounts at 1.5" away from center and 0 and 180 degrees, a LOC 36" chute at the top of the airframe, and 4 7" root 4" tip chord 5" span symmetrical trapezoidal fins made of 0.125" birch.

I added a 500 gram mass object in the center of the rocket hoping to change its size (like making it 20 meters long) to influence the moment of inertia of the rocket.

I also added two G138 motors, with the one in the 180 degree position motor tube delayed by 0, 0.25, 1 second, or not ignited at all.

Then I ran simulations and plotted the side profiles.

I cannot get the simulations to consistently have the rocket forced to arc over from the asymmetric thrust. At first, I tried only loading one motor but it went straight despite having an off-center motor mount. Next I tried adding a small A motor to the other tube, but it worked on some motor configurations but not others. Finally, I actually loaded the motors and changed ignitions and it worked... but inconsistently. Some times I start the program it works, and sometimes they all follow a basically perfectly vertical trajectory.

Can anyone else duplicate this?

JoePfeiffer, 2021-11-29:
Off-axis thrust is not modeled at present.  Definitely something to add to a future release.

CarVac, 2021-11-30:
Interesting, because it at least seemed to be working intermittently.

I'll just wait for now, then.

CarVac, 2022-01-12:
I did some looking in the code and it doesn't seem that hard to add; it "just" requires adding a way to compute the thrust moment about the CG and add it to the expressions in `calculateAcceleration` in RK4SimulationStepper.java.

JoePfeiffer, 2022-01-12:
Yeah, I'm pretty sure it'll be straightforward.  But there are other things to focus on right now.