openrocket/openrocket #1063: Incorrect subsonic Cdp nosecone interpolation

storm345, 2021-12-04:
https://github.com/openrocket/openrocket/blob/f3440bac6846fd60da5dcac76e772f0539541e69/core/src/net/sf/openrocket/aerodynamics/barrowman/SymmetricComponentCalc.java#L400-L401

The referenced lines do not seem to select the correct values of a and b for a 'smooth transition of Cd and it's deriative at the transonic boundary' like is described in the openrocket techdoc (http://openrocket.sourceforge.net/techdoc.pdf) section 3.4.3. Instead the values selected introduce a significant discontinuity.

The following would work to generate the correct values of a and b:

b = minMach * minTransonicMachCdpDerivative ./ (minTransonicCdVal-CdpMach0);
a = (minTransonicCdVal-CdpMach0) ./ minMach.^b;

This obviously will result in a not-insignificant change to the calculation of pressure drag and so will change simulation results. If somebody has good data which can be used to validate the effect of fixing this and if it is indeed improving the simulation result accuracy (Or if somehow the erroneous code gave better results) would be I think a good approach before making the change.

storm345, 2021-12-04:
[image]
[image]

Visual comparison of difference in pressure drag coefficient for an elliptical nosecone with fineness ratio 4.67 and 0 nosecone transition angle

JoePfeiffer, 2021-12-07:
Just starting to look at this -- you appear to be using several variables (minTransonicMachCdpDerivative and others) that I can't find in the code.  I can guess what they should be from their names; do you have the calculations to find them somewhere?

storm345, 2021-12-08:
I was doing that as more descriptive names of what the variables represent. minTransonicMachCdpDerivative is equivalent to 'minDeriv' in the code (Derivative at minimum point in lookup data), minMach is 'min' in the code, minTransonicCdVal is 'min' in the code, CdpMach0 is 'cdMach0'. Hopefully this helps

JoePfeiffer, 2021-12-08:
Ah, OK.  Thanks

SiboVG, 2022-01-11:
Thanks for having brought this to our attention @storm345!