openrocket/openrocket #1065: exporting rocket to RKT gives error

Austinthemighty, 2021-12-28:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

trying to export a rkt file

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NoClassDefFoundError: com/sun/istack/Pool
	at com.sun.xml.bind.v2.runtime.JAXBContextImpl$JAXBContextBuilder.build(JAXBContextImpl.java:1156)
	at com.sun.xml.bind.v2.ContextFactory.createContext(ContextFactory.java:165)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:297)
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:286)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:409)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:721)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:662)
	at net.sf.openrocket.file.rocksim.export.RocksimSaver.marshalToRocksim(RocksimSaver.java:45)
	at net.sf.openrocket.file.rocksim.export.RocksimSaver.save(RocksimSaver.java:65)
	at net.sf.openrocket.file.GeneralRocketSaver.saveInternal(GeneralRocketSaver.java:229)
	at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:154)
	at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:107)
	at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:68)
	at net.sf.openrocket.gui.main.BasicFrame.saveAsRocksim(BasicFrame.java:1424)
	at net.sf.openrocket.gui.main.BasicFrame.exportAction(BasicFrame.java:1382)
	at net.sf.openrocket.gui.main.BasicFrame.access$10(BasicFrame.java:1360)
	at net.sf.openrocket.gui.main.BasicFrame$10.actionPerformed(BasicFrame.java:521)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
	at java.desktop/java.awt.Component.processEvent(Component.java:6391)
	at java.desktop/java.awt.Container.processEvent(Container.java:2266)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.ClassNotFoundException: com.sun.istack.Pool
[... 505 lines cut ...]
501570  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
501571  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
501572  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
501573  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
501574  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
501575  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
501576  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
501577  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
501578  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
501579  78069.985 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
501580  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
501581  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
501582  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
501583  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
501584  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
501585  78069.986 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: com.sun.istack.Pool
501586  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
501587  78069.986 STDERR (SwingExceptionHandler.java:49) 	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
501588  78069.987 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:520)
501589  78069.987 STDERR (SwingExceptionHandler.java:49) 	... 58 more
501590  78069.987 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
501591  78069.987 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
501592  78072.528 USER   (SwingExceptionHandler.java:220) User requested sending bug report
501593  78072.529 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2021-12-28:
Hi, thanks for the bug report. Could you be so kind as to attach the .ork-file that you produced this error, so we can use it for debugging purposes?

Austinthemighty, 2021-12-29:
[Rockets.zip](https://github.com/openrocket/openrocket/files/7787165/Rockets.zip)
here you go

SiboVG, 2021-12-29:
I am weirdly unable to recreate this issue... Will park this issue for others or future-me to deal with later

Austinthemighty, 2021-12-31:
weird, i wonder if it has something to do with me running the latest code and launching from eclipse, but im still having this issue

SiboVG, 2022-01-01:
Do you still have the same issue when running OR on lower versions of Java? I noticed you use version 17, which OR technically doesn't support yet

SiboVG, 2022-01-01:
Also: I've noticed that sometimes running OR from an IDE (e.g. Eclipse) can produce errors which are not present when you run OR from a JAR-file. So that could also be the issue.

RogueXT, 2022-01-23:
I attached the exported files along with one of mine for you to take a gander at.

No issues that I came across. This would be with both Java 11 and Java 17

[Issue1065.zip](https://github.com/openrocket/openrocket/files/7921604/Issue1065.zip)

Austinthemighty, 2022-01-28:
i have to run java 7 on my computer for work, and recently we got a new system that requires java 8 so I'm now running both, but it seam like its an eclipse thing, as when i downloaded @SiboVG jar file it worked

SiboVG, 2022-01-28:
Alright, thanks for the feedback @Austinthemighty

closed 2022-01-28 by SiboVG