openrocket/openrocket #1067: Improve GitHub repository

SiboVG, 2022-01-03:
The OR-repo is very basic. This can weaken people's attention for and interest in (developing) OpenRocket. To quote from GitHub's '[Community management and operation](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/about-community-management-and-moderation)'-page: 

> "GitHub offers tools that help repository maintainers establish and enforce standards of behavior in their communities on GitHub.com."

We should start using these tools to ease OR's development and interest.

What I would like is first of all go through the README-file, check if additions are needed (maybe add a screenshot of the OR homepage of 'A simple model rocket' in the beginning to have a visual to attract new visitors to). Additionally, GitHub has some other useful files that you improve the repo:

1. [CODE_OF_CONDUCT](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-code-of-conduct-to-your-project): this is just a sign of maturity for the repository, instead of being in the Wild West without clearly defined rules.
2. [CONTRIBUTING](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors): this is a **very** important file IMO. It will include all the necessary information to create a clear issue report, a good pull request etc. Especially for people inexperienced with open-source development or development in GitHub, this will be a game changer.
3. [LICENSE](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-license-to-a-repository): just generally good practice to include under which OpenRocket operates.
4. [SUPPORT](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-support-resources-to-your-project): this file will ease the way for OR-users or -developers to ask for help. We could link the Slack-server here, the OR-forum etc.
5. [CITATION](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files): this would be helpful for people using OpenRocket in research and would want to cite OpenRocket (I know people can just quote Sampo's thesis, but anyway, would be nice to add the citation directly to the repo).

Lastly, to make our lives more easy, we can also create [issue and pull request templates](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/about-issue-and-pull-request-templates). This enforces structured issue reports and pull request, which sounds like heaven for any developer.

I suggest to put all the files I just mentioned in the '.github'-directory of the repo to keep our business clean.

I am going to put this as a beta-blocker for the simple reason that once we release the beta, we will hopefully attract some new people to the repo. Preparing the repo for them could spark new devs and testers to get involved + it will keep new issue report that will come from the beta clean.

I know, documentation, *barf*, but the OR-repo is in need for some modernization...

hcraigmiller, 2022-01-04:
With regard to the citation, there are two formats, APA and MLA, with the latter being the most commonly used in educational environments. Within each, there are akso two different styles, as follows:

**APA CITATION STYLE**

1. Niskanen, Sampo (2013). OpenRocket 2022.xx.xx Computer software. 2022. GitHub, Available from: https://github.com/openrocket.

_OR_

2. OpenRocket 2022.xx.xx Computer software. 2022. GitHub, Available from: https://github.com/openrocket.

**MLA CITATION STYLE**

3. Niskanen, Sampo, et al., developers. OpenRocket. Version 2022.xx.xx, Open Source JAVA, Month Day, 2022. Software. GitHub, github.com/openrocket.

_OR_

4. OpenRocket. Version 2022.xx.xx, Open Source JAVA, Month Day, 2022. Software. GitHub, github.com/openrocket.

My preference would be MLA style, leaning towards 3 to give credit to the original source.

SiboVG, 2022-01-10:
> While these improvements will be very welcome, they are improvements. I'm not clear on why this should be a beta blocker.

I see your concern of it not being an essential **functional** change, but IMO it is an important **organizational** change before the beta comes out. The beta release can attract new users and potential developers. Preparing a clean work environment for them to a) report clear issues and b) help with development is more than just an improvement. They're also changes that are almost de facto standard in open source repositories nowadays.

So in short: I just want this to be done before OR gets more attention and traction, as to not lose potential new developer's interest and to help ourselves with clear issues that beta-testers will post.

JoePfeiffer, 2022-01-11:
I guess I'm just concerned that this release has been so long in the making, and we're getting close, and I don't want to see it delayed.

neilweinstock, 2022-01-11:
I agree with Joe here.  This is a fair bit of work that is not required before beta.  Getting an influx of interested developers should be our worst problem.

SiboVG, 2022-01-12:
Okay, I postponed it to a release blocker. I will take up this task, as a small side-project during my breaks. Thanks for the feedback guys 😊.

commit 143308f9 (2022-02-11): Merge pull request #1091 from SiboVG/issue-1067

[fixes #1067] Improve GitHub repository
commit 264f7357 (2022-02-11): [fixes #1067] Add CODE_OF_CONDUCT file
commit 2923cb2d (2022-02-11): [fixes #1067] Update README
commit 348910b4 (2022-02-11): [fixes #1067] Include SUPPORT.md
I know this one is a bit basic, but it's a start
closed 2022-02-11 by JoePfeiffer