openrocket/openrocket #1070: Change <axial-offset> in ORK file back to <position>

neilweinstock, 2022-01-12:
As debated on Slack, here is my official request to change <axial-offset> back to <position> in the ORK file format.  This removes an unnecessary incompatibility between files saved by the old and new versions of OR.

As part of the fix, the new version should always _save_ using <position> but should _accept_ either <axial-offset> *or* <position> when reading files, so as not to break all the files that have been created with dev versions up until now.

hcraigmiller, 2022-01-12:
So as to be able to better discuss backward compatibility, I have reviewed the example designs, as well as many of my own.

Backward compatibility is not simply a matter of changing a single parameter (<axial-offset> back to <position>), but, rather, of changing many parameters. Using _A simple model rocket_ as an example, to preserve backward compatibility, the following parameters would need to be changed back to their prior labels:

axialoffset
angleoffset
radiusoffset
instancecount
instanceseparation

And, these parameters don’t even include any of the new features, which cannot be made backward compatible.

My finding is that preservation of backward compatibility is not only unachievable because of the new features, attempting backward compatibility would lock-in less than optimal terms in labels rather than allow changes which better reflect what the label actually does. The term “position”, for example, is less descriptive of the label’s function than “axial-offset” in this instance.

That said, I believe that to the extent that backward compatibility can be achieved, it would best be done by creating a “save as 15.03" file type option. This would allow the translation of new parameter labels to the old, without stunting the development and improvement of the current unstable and future versions.

My thoughts.

JoePfeiffer, 2022-01-12:
So my opinion is clear here as well as on slack, I'd much rather see the file format improved going forward even at the expense of backward compatibility.  I would not object to a "save as 15.03" option if someone wanted to create it, but I don't see it as particularly valuable.

SiboVG, 2022-01-13:
You'll have to excuse me, but could someone give an overview of the changes that were made that block backwards compatibility? I thought it was just 'position' that was changed to 'axial offset', but looking at @hcraigmiller's comment, it seems that there are more instances changed? Are they all just renames of previous tags or are the changes more complex? And what were the previous labels for @hcraigmiller's mentioned labels?

JoePfeiffer, 2022-01-13:
This is nothing resembling a real analysis, but may be a data point.  When I was working on PR#1068 (Fix 1063 -- correct subsonic nose cone drag interpolation) I created the simplest rocket with an ellipsoidal nose cone I could, so I could pull some data out of the simulation.  I just created a nose cone, body tube, and finset, made the nose cone an ellipse with desired fineness, and stuck a motor in it.  The .ork is attached 
[ellipse-nose-cone.ork.txt](https://github.com/openrocket/openrocket/files/7864925/ellipse-nose-cone.ork.txt)
Here's what I get when I open it with 15.03
[image]
If I click "OK" to have it proceed regardless, it does open correctly (except with no motor).

neilweinstock, 2022-01-15:
I compared the simple model rocket between 15.03 and recent unstable build.  Here are the differences I found:
1) <position type="bottom"> ==> <axialoffset method="bottom">  (applies to many elements)
2) <fincount> ==> <instancecount>  (applies to fins).  It defaults to 3, so in the Simple model rocket it doesn't make a difference. But a 4-finned rocket saved in Unstable will appear as a 3-finned rocket in 15.03.
3) <rotation> ==> <angleoffset method="relative"> (fins.  Not sure what the method=relative here does, or what the alternatives are if any).  Default is zero, which is usually but not always the preferred value.
4) Unstable adds <radiusoffset method="surface"> to fins, set to zero.  I don't understand how it applies to fins, and there isn't (as far as I know) any UI to change it.  It is used in pods.
5) Unstable adds a new element <ignitionconfiguration> among the motor configurations.  I don't see any equivalent in the 15.03 files.  I'm a little confused by this.

Items 1-3 will break compatibility between new and old OR.  I can't guarantee if there are others lurking as well, but those are undoubtedly the high-runners.  So if I had my druthers I would revert these three.  However, I understand that the idea of reverting 3 changes is less appealing than that of reverting just 1.  Obviously, though, just reverting the one wouldn't accomplish anything.  It looks like 3 or none.

Items 4 and 5 are a mystery, but don't seem to affect compatibility.

hcraigmiller, 2022-01-16:
With the findings of @neilweinstock above, and in light of the research of @wolsen (that in "the history of OpenRocketSaver.java it appears that it was _intended_ that the file be saved in the oldest compatible format available"), as I previously stated, **I believe that to the extent that backward compatibility can be achieved, it would best be done by creating a “save as 15.03" file type option**. 

Translating a design from current version backwards, rather than reverting labels, has at least three benefits:

1. The work on the current unstable can move forward without the delay of recoding and testing to ensure that the reversions didn't break something.
2. Developers are free to change/improve labels over time, rather than being forced to use labels that may no longer describe the label's purpose with adequate specificity or grouping.
3. In some instances, a new element can be converted to an older, but compatible element (such as converting rail buttons to launch lugs, without just losing the rail guides altogether).

The translation may also be able to extend to motors, rather than just giving "No motor with designation...." In all, _if backward compatibility is a primary goal_, ****save as** translation seems to be a much better option** that reversion.

My thoughts.

hcraigmiller, 2022-01-30:
@JoePfeiffer said "If we really have to maintain backward compatibility, my favored way would be to have the problematic elements under both their old and new names" and @neilweinstock replied "I am fin[e] with that approach".

It looks like we have a consensus on this beta blocker, unless someone disagrees.

closed 2022-02-11 by JoePfeiffer