openrocket/openrocket #1071: Terrible rocket breaks OpenRocket

CuboidRaptor, 2022-01-13:
I used the optimization thingy to make a rocket.
I got bored, so I tried to make it as unstable as possible.
However, I found that my rocket is so abundantly terrible that some value in the code turns into NaN,
and then OpenRocket throws an error.
Bug report is [here](https://pastebin.com/jnc0cA0q)

JoePfeiffer, 2022-01-14:
Please post the .ork of the rocket that fails (you'll need to rename it to something like rocket.ork.txt to get github to accept it), and give a more specific description of what steps you used to get to the exception -- what were you modifying to try to make it unstable?

JoePfeiffer, 2022-01-21:
Issue relates to 15.03; not nearly enough information to see what's happening; no response to request for more information.

closed 2022-01-21 by JoePfeiffer