openrocket/openrocket #1074: Crashes to desktop during file open and file save operations

billhnm, 2022-01-21:
---------- Bug report ----------

Include detailed steps on how to trigger the bug:

1. Any file open or file save activity causes OR to crash to desktop
2. File > open -- crashes to desktop
3. Opening an example file and attempting to "Save as" -- crashes to desktop
4. OS entry in log is incorrect -- I'm running Windows 11
5. Tried running as administrator and running under compatibility mode -- no help

What does the software do and what in your opinion should it do in the case described above:

Include your email address (optional; it helps if we can contact you in case we need additional information):

dr.bill.hanson@gmail.com

(Do not modify anything below this line.)
---------- System information ----------
OpenRocket version: 15.03
OpenRocket source: default
OpenRocket location: C:\Users\buffp\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
JOGL version: 2.1.5
Current default locale: en_US
System properties:
  app.preferences.id=net/sf/openrocket/startup
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1252
  file.encoding.pkg=sun.io
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\buffp\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
  java.class.version=52.0
  java.endorsed.dirs=C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\endorsed
  java.ext.dirs=C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\ext;C:\WINDOWS\Sun\Java\lib\ext
  java.home=C:\Users\buffp\AppData\Local\OpenRocket\runtime
  java.io.tmpdir=C:\Users\buffp\AppData\Local\Temp\
  java.launcher.path=C:\Users\buffp\AppData\Local\OpenRocket
  java.library.path=C:\Users\buffp\AppData\Local\OpenRocket\app;C:\Users\buffp\AppData\Local\OpenRocket
  java.runtime.name=Java(TM) SE Runtime Environment
  java.runtime.version=1.8.0_111-b14
  java.specification.name=Java Platform API Specification
  java.specification.vendor=Oracle Corporation
  java.specification.version=1.8
  java.vendor=Oracle Corporation
  java.vendor.url=http://java.oracle.com/
  java.vendor.url.bug=http://bugreport.sun.com/bugreport/
  java.version=1.8.0_111
  java.vm.info=mixed mode
  java.vm.name=Java HotSpot(TM) 64-Bit Server VM
  java.vm.specification.name=Java Virtual Machine Specification
  java.vm.specification.vendor=Oracle Corporation
  java.vm.specification.version=1.8
  java.vm.vendor=Oracle Corporation
  java.vm.version=25.111-b14
  jnlp.jogamp.tmp.cache.root=jln2193877024628317928
  line.separator=\u000d\u000a
  os.arch=amd64
  os.name=Windows 10
  os.version=10.0
  path.separator=;
  sun.arch.data.model=64
  sun.awt.enableExtraMouseButtons=true
  sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
  sun.boot.class.path=C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\resources.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\rt.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\sunrsasign.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\jsse.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\jce.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\charsets.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\lib\jfr.jar;C:\Users\buffp\AppData\Local\OpenRocket\runtime\classes
  sun.boot.library.path=C:\Users\buffp\AppData\Local\OpenRocket\runtime\bin
  sun.cpu.endian=little
  sun.cpu.isalist=amd64
  sun.desktop=windows
  sun.io.unicode.encoding=UnicodeLittle
[... 71 lines cut ...]
  60      2.003 VBOSE  (DoubleModel.java:890) DoubleModel[RocketFigure:Rotation] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@51f8e357
  61      2.178 DEBUG  (AbstractAerodynamicCalculator.java:98) Voiding the aerodynamic cache
  62      2.178 DEBUG  (AbstractMassCalculator.java:36) Voiding the mass cache
  63      2.178 VBOSE  (Rocket.java:357) Added ComponentChangeListener net.sf.openrocket.gui.scalefigure.RocketPanel$4@5a2c4d08, current number of listeners is 10
  64      2.178 DEBUG  (JarUtil.java:40) Found jar file using codeSource
  65      2.178 DEBUG  (ExampleDesignFile.java:115) Current jar file is: C:\Users\buffp\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
  66      2.203 VBOSE  (Rocket.java:357) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@5da57798, current number of listeners is 11
  67      2.203 VBOSE  (Rocket.java:357) Added ComponentChangeListener net.sf.openrocket.gui.main.BasicFrame$1@6f3246b7, current number of listeners is 12
  68      2.240 DEBUG  (UpdateInfoRetriever.java:173) Update response code: 404
  69      2.240 WARN   (UpdateInfoRetriever.java:184) Unknown server response code: 404
  70      2.240 DEBUG  (BasicFrame.java:265) BasicFrame instantiation complete
  71      2.272 INFO   (SwingStartup.java:194) Checking update status
  72      2.272 INFO   (SwingStartup.java:83) Startup complete
  73      2.350 DEBUG  (GUIUtil.java:310) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1728,height=972]
  74      2.350 DEBUG  (GUIUtil.java:310) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1728,height=972]
  75      2.365 DEBUG  (GUIUtil.java:310) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1728,height=972]
  76     11.116 USER   (BasicFrame.java:762) Debug log selected
  77     11.116 INFO   (DebugLogDialog.java:116) Adding log listener
  78     38.805 USER   (DebugLogDialog.java:361) Closing debug log dialog
  79     38.805 INFO   (DebugLogDialog.java:364) Removing log listener
  80     38.805 DEBUG  (GUIUtil.java:235) Clearing all models of window net.sf.openrocket.gui.dialogs.DebugLogDialog[dialog0,203,231,1574x745,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title=OpenRocket debug log,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,4,27,1566x714,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
  81     38.805 DEBUG  (MemoryManagement.java:56) Adding object into collectable list: net.sf.openrocket.gui.dialogs.DebugLogDialog[dialog0,203,231,1574x745,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title=OpenRocket debug log,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,4,27,1566x714,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
  82    228.571 USER   (BasicFrame.java:749) Bug report selected
  83    228.571 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

hcraigmiller, 2022-01-21:
First, are you running under Windows 11 Home or Windows 11 Pro?
Second, please open the "OpenRocket 15.03 Properties" tab and provide the "Target" text string and the "Start in" text string.
Third, did you attempt to customize the installation process in any way (or are you running the .jar file)?

billhnm, 2022-01-21:
Hi, 

 

I’m running Win 11 Pro.

 

Target: C:\Users\buffp\AppData\Local\OpenRocket\OpenRocket.exe

Start in: C:\Users\buffp\AppData\Local\OpenRocket

 

The installer offered no option to customize.  It’s all in the default location.  

This behavior was running OpenRocket.exe – I did try running the .jar file (which I had been running previously under Win 10 Pro) but that just shows the splash screen and immediately crashes.  

 

 

Best,

- Bill H

SiboVG, 2022-01-21:
Hi Bill, could you check if Windows' Event Viewer application shows anything special after OR crashes, like explained here: https://www.cyberlink.com/support/faq-content.do?id=10449#:~:text=find%20event%20logs%3A-,Windows,-7%3A.

billhnm, 2022-01-21:
Here are two event logs: 

 

Crash on file > open

 

Log Name:      Application

Source:        Application Error

Date:          1/20/2022 10:29:00 AM

Event ID:      1000

Task Category: (100)

Level:         Error

Keywords:      Classic

User:          N/A

Computer:      DESKTOP-8019MKJ

Description:

Faulting application name: OpenRocket.exe, version: 0.0.0.0, time stamp: 0x57e43091

Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000

Exception code: 0xc0000005

Fault offset: 0x0000000000000000

Faulting process id: 0x41d4

Faulting application start time: 0x01d80e23366718e5

Faulting application path: C:\Users\buffp\AppData\Local\OpenRocket\OpenRocket.exe

Faulting module path: unknown

Report Id: 1306bde5-b31d-49cf-88da-78eebbaf87d2

Faulting package full name: 

Faulting package-relative application ID: 

Event Xml:

<Event xmlns=http://schemas.microsoft.com/win/2004/08/events/event>

  <System>

    <Provider Name="Application Error" />

    <EventID Qualifiers="0">1000</EventID>

    <Version>0</Version>

    <Level>2</Level>

    <Task>100</Task>

    <Opcode>0</Opcode>

    <Keywords>0x80000000000000</Keywords>

    <TimeCreated SystemTime="2022-01-20T17:29:00.7837296Z" />

    <EventRecordID>22621</EventRecordID>

    <Correlation />

    <Execution ProcessID="4304" ThreadID="0" />

    <Channel>Application</Channel>

    <Computer>DESKTOP-8019MKJ</Computer>

    <Security />

  </System>

  <EventData>

    <Data>OpenRocket.exe</Data>

    <Data>0.0.0.0</Data>

    <Data>57e43091</Data>

    <Data>unknown</Data>

    <Data>0.0.0.0</Data>

    <Data>00000000</Data>

    <Data>c0000005</Data>

    <Data>0000000000000000</Data>

    <Data>41d4</Data>

    <Data>01d80e23366718e5</Data>

    <Data>C:\Users\buffp\AppData\Local\OpenRocket\OpenRocket.exe</Data>

    <Data>unknown</Data>

    <Data>1306bde5-b31d-49cf-88da-78eebbaf87d2</Data>

    <Data>

    </Data>

    <Data>

    </Data>

  </EventData>

</Event>

 

Crash on file > save as

 

Log Name:      Application

Source:        Application Error

Date:          1/20/2022 10:33:08 AM

Event ID:      1000

Task Category: (100)

Level:         Error

Keywords:      Classic

User:          N/A

Computer:      DESKTOP-8019MKJ

Description:

Faulting application name: OpenRocket.exe, version: 0.0.0.0, time stamp: 0x57e43091

Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000

Exception code: 0xc0000005

Fault offset: 0x0000000000000000

Faulting process id: 0x55d8

Faulting application start time: 0x01d80e23ca258fea

Faulting application path: C:\Users\buffp\AppData\Local\OpenRocket\OpenRocket.exe

Faulting module path: unknown

Report Id: ce5684fb-93e8-41a2-8f90-c374c63ff8c7

Faulting package full name: 

Faulting package-relative application ID: 

Event Xml:

<Event xmlns=http://schemas.microsoft.com/win/2004/08/events/event>

  <System>

    <Provider Name="Application Error" />

    <EventID Qualifiers="0">1000</EventID>

    <Version>0</Version>

    <Level>2</Level>

    <Task>100</Task>

    <Opcode>0</Opcode>

    <Keywords>0x80000000000000</Keywords>

    <TimeCreated SystemTime="2022-01-20T17:33:08.6393411Z" />

    <EventRecordID>22625</EventRecordID>

    <Correlation />

    <Execution ProcessID="22396" ThreadID="0" />

    <Channel>Application</Channel>

    <Computer>DESKTOP-8019MKJ</Computer>

    <Security />

  </System>

  <EventData>

    <Data>OpenRocket.exe</Data>

    <Data>0.0.0.0</Data>

    <Data>57e43091</Data>

    <Data>unknown</Data>

    <Data>0.0.0.0</Data>

    <Data>00000000</Data>

    <Data>c0000005</Data>

    <Data>0000000000000000</Data>

    <Data>55d8</Data>

    <Data>01d80e23ca258fea</Data>

    <Data>C:\Users\buffp\AppData\Local\OpenRocket\OpenRocket.exe</Data>

    <Data>unknown</Data>

    <Data>ce5684fb-93e8-41a2-8f90-c374c63ff8c7</Data>

    <Data>

    </Data>

    <Data>

    </Data>

  </EventData>

</Event>

billhnm, 2022-01-22:
> Thanks! Okay, so both errors have exception code '0xc0000005', which indicates a memory access violation. This does makes sense since the program crashes when trying to open a file or saving a file, which both require memory access.
> @billhnm does the program still crash if you open it as an administrator (e.g. look up the program in windows search, right click the application icon and then select 'Run as administrator')?
> If that does not work, you can try to go over some of the steps explained here: https://www.ionos.co.uk/digitalguide/server/configuration/resolving-0xc0000005-errors/.

Hi,

On the Rocketry Forum, Reinhard suggested that having the “God Mode” shortcut (which I did) could have been the problem.

Sure enough, installing OR on a laptop without God Mode and then removing God Mode from my desktop also fixed the problem.  OR works fine on both machines now.

I had previously run it as Administrator and that didn’t help.  Looks like God Mode was indeed the culprit.

I hope this helps for future iterations of the program.

Best,
Bill H

SiboVG, 2022-01-22:
Ah, great to hear, thanks for the update!

closed 2022-01-22 by SiboVG