openrocket/openrocket #1075: Opened attached ORK file, OpenRocket crashed, bug report and ORK files attached

RogueXT, 2022-01-24:
I opened the attached file, clicked save. Dialog pops up asking to save all data or just the rocket itself. Clicked sav
[Estes Vapor.zip](https://github.com/openrocket/openrocket/files/7929175/Estes.Vapor.zip)
e all data.

The ORK file that is separate is the before and the ORK in the Bug Report is the after.

[Estes-Vapor-BugReport-20220124-0510pm.zip](https://github.com/openrocket/openrocket/files/7929170/Estes-Vapor-BugReport-20220124-0510pm.zip)

hcraigmiller, 2022-01-24:
Please provide the date on which you downloaded this version of the unstable OpenRocket, and confirm the version of Java that you are using (the unstable is designed for use with Java 11 at this point).

RogueXT, 2022-01-25:
I downloaded the source from GitHub on January 22 @12:36PM. I compiled/built the JAR file myself. This particular build is using 11.0.13 from Oracle’s website.

 

JP

hcraigmiller, 2022-01-25:
Can you download the .jar file attached to PR #1077 and test your design using that .jar to see if that works?

RogueXT, 2022-01-25:
Yuppers, it works just fine now.

 

-JP

 

 

Groovy Regards,

 

 

JP Johnston ME CSWP

Portland State Aerospace Society

Rogue Experimental Technologies | JP Parametrics

Email:  ***@***.***> ***@***.*** |  ***@***.***> ***@***.*** |  ***@***.***> ***@***.***

Mobile: 260.431.3664

Home: 260.440.1483

Portfolio:  <https://grabcad.com/jp.johnston.me.cswp-1/models> GrabCAD

LinkedIn:  <https://www.linkedin.com/in/roguextech/> CV/Profile

JoePfeiffer, 2022-01-25:
Hmmm, I can't think of anything I did in that PR that should have affected this...

I built both from current unstable and from PR-1077 and didn't see the crash either time (note that I saw a slightly different set of options for saving than you reported -- two options for simulation data and "primary figures only". I'm assuming that's the one you picked?)

hcraigmiller, 2022-01-25:
I believe that it may likely be related to the building of the .jar file by @RogueXT. I remember having to try several methods myself before I got a fully functioning .jar.

@RogueXT, once you are satisfied that the .jar file referred to above is fully functional, please close this issue.

closed 2022-01-26 by RogueXT