openrocket/openrocket #1076: "favorite" preset components dropdown doesn't update until edit window is reopened.

JoePfeiffer, 2022-01-25:
To duplidate:
1) Select a component to edit, and open the edit window
2) Open the dropdown in the upper right corner. Observe the selected presets
3) Select "From database..." to open the component preset chooser dialog
4) Change the set of selected components, either adding or deleting, or both
5) Click "OK" to close the dialog, and open the dropdown again. Observe that the selected presets is unchanged
6) Close and reopen the edit window.  When the dropdown is reopened, the changes have taken place.

hcraigmiller, 2022-01-25:
I am unable to reproduce this problem using _A simple model rocket_. If I change the nose cone to one from the database, the _Select preset_ window immediately changes to the part number of the database component when the _Okay_ button in pushed.

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

JoePfeiffer, 2022-01-25:
Here's what I'm seeing.

First photo, dropdown is open and I'm about to select "From database..."
[image]

Second photo, I'm in the component preset choose dialog.  I've sorted by the first column, deselected, the BNC-20B, and selected the BNC20L (BNC-20B doesn't appear in the image due to sorting).
[image]

Third photo, I've clicked OK and reopened the dropdown. BNC-20B still appears, BNC20L does not.
[image]

Fourth photo, I've closed and reopened the edit window, and opened the dropdown.  Correct presets are present.
[image]

hcraigmiller, 2022-01-25:
Error replicated in Windows 11.

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

SiboVG, 2022-01-31:
Also would like to point out that clicking 'Cancel' in the 'Choose component preset' window does not undo selections that you have made, there is just no difference between the 'Ok' and 'Cancel' button. If I select 2 new presets and click 'Cancel', the presets should not be added to the preset dropdown, as is the case currently.

JoePfeiffer, 2022-01-31:
Yes.  I think what will work most smoothly with the rest of the component editor will be to replace the "OK" and "Cancel" buttons with a single "Close" button.

commit 1ae589d8 (2022-02-01): [fixes #1076] Use Close button instead of Ok/Cancel
commit 3541899b (2022-02-01): [fixes #1076] Move PresetModel database listener internally
setComponent() of ComponentConfigDialog called pack(), which triggered invalidate() in RocketComponentConfig, thus removing presetModel from the database listeners before the Component Preset window could even be called
commit 620b450e (2022-02-01): [fixes #1076] Fix layout issues ComponentPresetChooserDialog
commit 8e50f520 (2022-02-01): Merge pull request #1103 from SiboVG/issue-1076

[fixes #1076] Fix preset combobox not updating
closed 2022-02-01 by JoePfeiffer