openrocket/openrocket #1078: Add Opacity slider to Appearance panel

neilweinstock, 2022-01-25:
Unstable release adds support for alpha channel in the 3D renders.  Right now the only way to control it is inside the color picker, in the RGB/HSL/blah blah blah section.  This is far too hidden and too inaccessible.  Propose a slider, labeled "Opacity", somewhere under the color button, either above or below the "Shine" control.

The Alpha channel controls in the color picker could still be there, it doesn't hurt to have it in both places.  Obviously, editing in either place should show up in the other.

hcraigmiller, 2022-01-25:
The transparency slider needs to be added to the RGB and CMYK color picker pages (see Issue #401).

neilweinstock, 2022-01-25:
There's already an alpha channel setting there.  It works and is fine.  No need to go mucking with the color picker.  Purpose here is simply to surface the same functionality in a more friendly way, e.g.:

[image]

I would really prefer the opacity slider to go from 0-100%, but it may be much easier to just have it be 0-255 and therefore be exactly the same as the alpha channel value.  Otherwise need to convert back and forth, which is possible but not idea.

hcraigmiller, 2022-01-25:
I agree that the opacity (transparency) slider should go from 0-100% as suggested by @neilweinstock.

neilweinstock, 2022-01-25:
I may try to implement this myself at some point if no one else gets to it first.

PR #1197 (2022-02-28): [fixes #1078] Add opacity slider to appearance config
This PR fixes #1078 by adding an opacity slider to the appearance config dialog:

https://user-images.githubusercontent.com/11031519/156026176-9c48d7d9-eb94-4b14-a869-e1e2eb287cae.mov

(I choose to use the term 'Opacity' instead of 'Transparency'. I know that the color picker has 'Transparency', but 'Opacity' is more common in all the programs I've worked with so far).

Here is a [jar file](https://github.com/openrocket/openrocket/suites/5477442873/artifacts/174667095) for testing.
commit 6ca76c4e (2022-02-28): [fixes #1078] Add opacity slider to appearance config
commit a6ddbcd1 (2022-02-28): Merge pull request #1197 from SiboVG/issue-1078

[fixes #1078] Add opacity slider to appearance config
closed 2022-02-28 by JoePfeiffer