openrocket/openrocket #1079: "License" and "about" popups and databases

JoePfeiffer, 2022-01-25:
Since we will be distributing Dave's parts database under the Apache license, presumably the "License" popup is the right place to declare that?  And do we need to include the whole license in the popup, or is it acceptable to point back to the original text at https://www.gnu.org/licenses/gpl-3.0.en.html?  Same question for Apache.

Also, presumably his database and thrustcurve should be mentioned in the "about" popup.

hcraigmiller, 2022-01-25:
The "License" pop-up should contain a brief statement about each license **in addition to** (1) an internal link to the full text of each license that is contained within a separate file _and_ (2) an internal link to the license on the holder's site.

For example:

**GNU GENERAL PUBLIC LICENSE**

OpenRocket - A model rocket simulator
Copyright 2007-2022 Sampo Niskanen _et al._

This program is free software: you can redistribute it and/or modify it under the terms of the GNU General Public License [link to full text License file] as published by the Free Software Foundation, either version 3 of the License, or any later version.

This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for more details.

A copy of the GNU General Public License may be viewed here [link to full text License file] You may also obtain a copy of the License at:

    https://www.gnu.org/licenses/. 

OpenRocket developers may be contacted electronically at: 

    openrocket-devel@lists.sourceforge.net
    https://github.com/openrocket

**APACHE LICENSE**

OpenRocket - A model rocket simulator
Copyright 2007-2022 Sampo Niskanen _et al._

Licensed under the Apache License, Version 2.0 (the "License"); you may not use this product except in compliance with the License [link to full text License file]. You may also obtain a copy of the License at:

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software distributed under the License is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the License [link to full text License file] for the specific language governing permissions and limitations under the License.

**It is also recommended that the notices be included at the top of specific associated files (such as the parts library).**

JoePfeiffer, 2022-01-25:
This looks like OR is being distributed under both GPL and Apache.  Should the top of the Apache license read

APACHE LICENSE

openrocket-database - Enhanced components database for OpenRocket
Copyright [Dave's Information]

Licensed under the Apache License, Version 2.0 (the "License"); you may not use this work except in compliance with the License [link to full text License file]. You may also obtain a copy of the License at:

Since (as far as I know) the license files aren't being bundled up in the jar file, do the "internal" links go back to github?  As you can tell, I know *nothing* about doing this stuff right!

hcraigmiller, 2022-01-25:
Good catch. 

APACHE LICENSE

Enhanced components parts database for OpenRocket
Copyright [Dave's Information]

*  *  *

**The license files should be bundled in (or with) the .jar file**, as an internal link from the License pull-down. To clarify, in the Apache License example in my prior comment, the phrase "the **_License_**" that is both bold and italic, when clicked, would bring up the full text of the Apache License within OpenRocket. In addition, clicking on the http link would take you to the Apache License page on the Apache website.

JoePfeiffer, 2022-01-25:
Hmm, looks like our LICENCE.TXT actually is bundled in the .jar file at present, and there is a way (that I've only started to explore) to set a URL to it in the program.  Adding Dave's license as well looks straightforward.

SiboVG, 2022-01-28:
In PR #1082, I used the commonmark-java library which falls under the BSD-2-Clause License with following license info:
```
Copyright (c) 2015-2016, Atlassian Pty Ltd
All rights reserved.

Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions are met:

* Redistributions of source code must retain the above copyright notice, this
  list of conditions and the following disclaimer.

* Redistributions in binary form must reproduce the above copyright notice,
  this list of conditions and the following disclaimer in the documentation
  and/or other materials provided with the distribution.
```

So I guess the copyright also needs to be included somewhere?

hcraigmiller, 2022-02-05:
If I understood what the intellectual property attorney said, the License Pop-Up should look something very close to the attached .pdf

[OpenRocket_License.pdf](https://github.com/openrocket/openrocket/files/8007168/OpenRocket_License.pdf)

Once all of the licensed elements are identified, I can run it by the attorney, if that is acceptable.

closed 2022-02-10 by SiboVG