openrocket/openrocket #1080: 3D view blacks out while resizing (M1 Mac)

neilweinstock, 2022-01-26:
On my M1 iMac, if any 3D view is enabled, then the entire rocket display blacks out while resizing.  As soon as the mouse stops moving, the display reappears.

This is more of an annoyance than anything.

hcraigmiller, 2022-01-26:
In Windows, decreasing the window size does not result in any portion of the window blacking out. But, if the window size is rapidly increased, the portion of the screen being brought into the window is black until the expansion catches up with the mouse. It seems to increase in blocks that increase at different rates, the component window increases at a different rate than the design window; both leave a slight black edge until the mouse is actually released.

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

SiboVG, 2022-01-26:
Same blackout behavior on my M1 MacBook Pro.

RemingtonHolder, 2022-03-22:
Should this issue be merged with #939

neilweinstock, 2022-03-22:
It is unclear if all the 3D issues should be smooshed together into one.  It probably doesn't matter, because the solution is going to be replacing JOGL rather than targeted bug fixes.

closed 2026-08-24 by SiboVG