openrocket/openrocket #1108: [Bug] 'Simulation extensions and scripting' example design throws exception when simulating

SiboVG, 2022-02-06:
The 'Simulation extensions and scripting' example design throws an exception when running the second/third simulation.
[Bug report Sim3 - Roll control.txt](https://github.com/openrocket/openrocket/files/8011245/Bug.report.Sim3.-.Roll.control.txt)
[Bug report Sim2 - Active roll control.txt](https://github.com/openrocket/openrocket/files/8011243/Bug.report.Sim2.-.Active.roll.control.txt)

Tested on latest unstable, macOS version 12.1, MacBook Pro M1.

hcraigmiller, 2022-02-06:
> The 'Simulation extensions and scripting' example design throws an exception when running the second/third simulation. [Bug report Sim3 - Roll [...]

When attempting to open either of the two design files, I get an error message.
[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

SiboVG, 2022-02-06:
> > The 'Simulation extensions and scripting' example design throws an exception when running the second/third simulation. [Bug report Sim3 - Roll control.txt](https://github.com/openrocket/openrocket/files/8011245/Bug.report.Sim3.-.Roll.control.txt) [Bug report Sim2 - Active roll control.txt](https://github.com/openrocket/openrocket/files/8011243/Bug.report.Sim2.-.Active.roll.control.txt)
> 
> When attempting to open either of these two design files, I get an error message.

Yes, they're not design files, but the bug reports after running simulation 2 and simulation 3 🙃

SiboVG, 2022-02-08:
I don't have this issue when I run OR from my IDE IntelliJ, only when I run it from a JAR.

PR #1507 (2022-08-02): Resolving issue #1270, unexpected exception with simulation scripting
This PR fixes #1270, fixes #308, fixes #1108 and fixes #826.

As the Nashorn scripting engine is deprecated, as part of the resolution this replaces Nashorn with the Graal.Js scripting engine.  This includes the addition of the following dependencies

graal-sdk-22.1.0.1.jar
icu4j-71.1.jar
js-22.1.0.1.jar
js-scriptengine-22.1.0.1.jar
truffle-api-22.1.0.1.jar
closed 2022-08-02 by SiboVG