openrocket/openrocket #1121: Change Request: Hide motors not in production

hcraigmiller, 2022-02-12:
When opening the "Select a rocket motor" dialog box, most users expect the motors shown to be in regular production. Current;ly, the default state is to show OOP motors.

This request is to change the hide motors not in production by either:
1. Making "Hide motors which are not in regular production" the default setting when the "Select a rocket motor" dialog box is opened; or
2. Rewording "Hide motors which are not in regular production" to "Show motors which are not in regular production", with the default being off, similar to the legacy component library functionality.

I prefer number 2.

JoePfeiffer, 2022-02-12:
I definitely prefer #2.  Negative options (turn this option on to not see/do something) are a pet peeve of mine.

hcraigmiller, 2022-03-22:
Current unstable. Option 1 above has been implemented. 

[image]

Should this issue be closed, or is a change to option 2 anticipated?

closed 2022-03-22 by JoePfeiffer