openrocket/openrocket #1135: Get file associations working correctly on packaged installers

neilweinstock, 2022-02-18:
Right now, file association does not seem to work correctly in the packaged installers.  It is unclear what the problem is right now.

This would be a good thing to get working before the final release.

JoePfeiffer, 2022-02-21:
Putting this here so it doesn't get lost -- here's a discussion on stackexchange suggesting MacOS file associations may need some code in OR itself.  Haven't investigated carefully.  https://stackoverflow.com/questions/67848790/how-to-pass-a-path-from-open-with-to-an-executable-created-by-install4j

neilweinstock, 2022-02-22:
> Putting this here so it doesn't get lost -- here's a discussion on stackexchange suggesting MacOS file associations may need some code in OR itself. Haven't investigated carefully. https://stackoverflow.com/questions/67848790/how-to-pass-a-path-from-open-with-to-an-executable-created-by-install4j

Well that is rather on-the-nose.  Hopefully implementation is not too hard.  I would guess that this also adds support for dragging a file onto the OR dock icon.

One more request, if possible (I don't know what would be involved):  In the case where the OR app is invoked by double-clicking an ORK file, then OR should *not* also open the last opened file.  I believe this behavior should apply to other OSes as well.

jimmiedave, 2022-02-27:
Doesn't appear to work on Ubuntu 20.04 here. .ork files show up as type "binary" in the file explorer, and if I double-click, they open with an archive expander.

Not certain whether I'd already associated .ork files to the expander in tinkering with them before (looking at a different bug months ago) or not, but if we're to offer making a file type association, we probably need to check for and change (or force-change) existing associations (one other case for that might be some user proactively attempting to associate .ork with 15.03 and forgetting she'd done so. Another might be associating to some future version N of OpenRocket, then breaking them when N++ is released).

JoePfeiffer, 2022-02-27:
Ah, you are correct.  I was fooling myself that it was working under Debian; it's not.

jimmiedave, 2022-02-28:
Tested on my Win7 VM just now. Double-clicking .ork file (after I'd asked installer to associate .orks with OR) results in a "Windows can't open this file" app chooser dialog.

JoePfeiffer, 2022-02-28:
To see if I was also fooling myself on my Win10 machine, try a reboot and see if the association starts working (it did for me, though I'm told it shouldn't have been necessary).

PR #1432 (2022-06-14): Revised installer configuration to enable desktop icon and file associations
The major change is it turns out you need to do the file installation before trying to create the integrations.

Installers to test in https://drive.google.com/drive/folders/1xaUZXnl9q6CDG-CbkqIQqpFrBz63Gw2k?usp=sharing

Fixes #1135
closed 2022-06-14 by SiboVG