openrocket/openrocket #1137: Homogenize capitalization of rocket component names

SiboVG, 2022-02-19:
This is a very nitpicky thing, I know, but it bothers me slightly that there are some rocket components that by default have all the words in their name capitalized, while others don't.

More specifically, I would make the following changes:

- Inner Tube --> Inner tube (in the component tree)
- Launch Lug --> Launch lug (both in the component tree and the create new component button)
- Rail Button --> Rail button (both in the component tree and the create new component button)
- Mass Component --> Mass component (both in the component tree and the create new component button)

This would just tidy things up a bit.

neilweinstock, 2022-02-19:
I agree with your proposal.  First word capitalized only.

JoePfeiffer, 2022-02-19:
I'd actually prefer the opposite -- all words capitalized.

SiboVG, 2022-02-19:
Right, I'll post a practical comparison of the two here in a sec.

SiboVG, 2022-02-19:
[image]

I must say, seeing it in real, I prefer the capitalized version.

neilweinstock, 2022-02-19:
That's fine with me.  I think first word capitalized is likely more "correct", but capitalization rules can be funny in this sort of situation.

Consistency is most important.  Go for the capitalized version.

hcraigmiller, 2022-02-19:
Either works for me, no real preference.

RogueXT, 2022-02-20:
I Vote for all Capital Myself. It is a Very Bad Habit I have Developed in Forty (40) Years of Designing Tooling. :-)

-JP

PR #1142 (2022-02-21): [fixes #1137] Use capitalized component names
This PR fixes #1137 by capitalizing rocket component names. Additionally, I noticed that the Body Tube, Nose Cone and Transition had a specific name in their config dialog to change the length ('Body tube length', 'Nose cone length' and 'Transition length'). Since these are redundant and other components also just use 'Length', I changed the translations of those labels to just 'Length'.
commit 12cefe09 (2022-02-21): [fixes #1137] Use capitalized component names + Body, Nose & Transition Length --> Length
commit 88c8d7fe (2022-02-21): Merge pull request #1142 from SiboVG/issue-1137

[fixes #1137] Use capitalized component names

Looks good, merging.
closed 2022-02-21 by neilweinstock