openrocket/openrocket #1144: Get signing keys and update install4j license

JoePfeiffer, 2022-02-20:
We currently have a license for install4j version 8.  As version 9 is out, we should apply for a new license (the process looks easy, and we've already complied with their condition of advertising them on our site).

We should also investigate getting signing keys for Windows and MacOS, so our users don't get security warnings when they download and install the packages.  I've got *no* idea what the process is for this.

RogueXT, 2022-02-20:
I have no idea about the process either however, I did find these links for [Windows](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/managing-the-digital-signature-or-code-signing-keys)and [Mac](https://developer.apple.com/support/code-signing/).

Just trying to assist. I will also ask our software guru if he knows anything about this process tomorrow. If he does, I will see if he can guide me thru the process a little and then hand it off to whomever it needs to go to. If that is okay with y'all...

-JP

neilweinstock, 2022-02-21:
At least for the Mac, there needs to be an Apple Developer account through which to sign the app.  I am investigating where it is OK to use mine (which is connected to my job) or whether it needs to be someone else's.

The Install4J docs make some cryptic references to the need for app signing being connected to the form of the installer, but this makes no sense to me... *any* app should need to be signed in order to able to run without needing to go through extra security approvals.

SiboVG, 2022-02-22:
> We currently have a license for install4j version 8. As version 9 is out, we should apply for a new license (the process looks easy, and we've already complied with their condition of advertising them on our site).

I will make a request for it. They ask you to enter the number of developers (stated on the submission form as 'core committers of a project'), so I put the number of current core-developers/admins: 13.

neilweinstock, 2022-02-24:
I marked this down as a Release Blocker, but if we can't figure out a way to get it done then we'll defer.  But it's worth a try.

JoePfeiffer, 2022-02-24:
Should be mentioned Sibo's request for an install4j version 9 license was successful.

closed 2023-01-13 by neilweinstock