openrocket/openrocket #1145: Allow pods to be attached to fins and tube fins

neilweinstock, 2022-02-21:
This feature was discussed and then deferred from this release, I wanted to make sure it was entered here to keep track of it.

Presently, pods must be attached to body tubes.  This is necessary but not sufficient functionality.  If the contents of the pods are intended to be attached to the ends of fins, then by far the most sensible way to enter them is, indeed to attach them to fins.

In this case, the parameters need to change a bit (this is where implementation perhaps becomes a bit delicate).  Can the "General" tab in Pod Set configuration change behavior depending on the type of parent?  I don't know.

1) "Radius positioning method" should, by default, be to attach to the end of the fin.  Radial distance would then be an offset from that.  I'm not sure what you would call it.  It is most similar to "Position relative to the component surface", but that's not really the correct terminology.  Arguably, even "Radius Positioning Method" isn't really the correct name for this parameter when it's connected to fins.
2) "Angle" and "Number of Copies" do not apply.  there would automatically be one pod attached to each fin.

Other parameters stay the same.

neilweinstock, 2022-05-12:
I am wondering if the pod that attaches to a fin should be a separate component from the existing Pod Set.  The fin pod would automatically be one for each instance of the fin, rather than a set with its own instancing.

JoePfeiffer, 2022-05-12:
I think I'd have to have a better sense of just what RS does with pods to have an opinion.