openrocket/openrocket #1146: Show pods on rocket figure view

neilweinstock, 2022-02-21:
Right now pods are invisible.  It can be hard to figure out if you're putting them in the right place.  It would be nice to have a small mark on the rocket figure view (forget the 3D view) to show where they are.  I do not know exactly what those marks should look like, something to think about.

neilweinstock, 2022-07-11:
A note would be better than nothing but I already know that they're invisible... I want to see where they are.  I'm still unsure exactly what they should look like, though, or what size they should be.  It would probably be best if they were always the same size, regardless of the magnification of the rocket.

jimmiedave, 2022-07-11:
I'd like the same as you, but there's currently no indication that OpenRocket isn't broken when you add a pod set - because they're invisible. No success message, no reminder that they won't appear, and of course, no presence in the model views.

A natural reaction is to think there's something wrong.

neilweinstock, 2022-07-11:
I suppose I must agree. Until we display the pods in the rocket figure, a message in the config dialog seems like a good idea.

But we should really just display them in the rocket figure. 😀

jimmiedave, 2022-07-11:
Interesting - so pods are basically a virtual component (almost like a fin set, but with no fins) - is that what I'm understanding now? As such, they'd _have_ no appearance. I had no idea that's how it worked.

Well, perhaps a note to _that_ effect, if I have it right. Your idea of a line makes sense, Joe - demonstrating the hierarchy - but until this moment I had no idea that I was thinking completely wrong about this.

neilweinstock, 2022-07-11:
Correct. Pods are unique within the program in that respect.

I like Joe’s idea.  For the markings, I’d use a small circle in a unique color (purple maybe?). It should show as the same circle from all angles, which technically would make it a sphere. I also would want them always to render at the same size.

I suppose a mock-up or two would be helpful here. I’ll try to throw something together tonight.

SiboVG, 2022-07-11:
I also like Joe's suggestion. I would however use a cross for both the side view and back view (not that it has any significance, but this would be analogous to an "empty" object in the Blender 3D program - see the 3D visualization in the image below).
[image]

neilweinstock, 2022-07-11:
A little plus sign would be good, sure.

[image]<https://user-images.githubusercontent.com/11031519/178375577-31d56a84-9085-4fcc-8da9-df8b7daac996.png>

JoePfeiffer, 2022-07-11:
> Interesting - so pods are basically a virtual component (almost like a fin set, but with no fins) - is that what I'm understanding now? As such, they'd _have_ no appearance. I had no idea that's how [...]

Much more so than a fin set -- you can't have a fin set with 0 fins, so they always have some physical reality and thus appearance.

More like a stage or a parallel booster set, or even the rocket itself.  When you first open OR, the parts tree shows the Rocket at the root of the tree, and the Sustainer as its sole child.  The design window is empty. If you do something like select the sustainer in the parts tree and add a Stage, you now have the Rocket, Sustainer, and Stage in the parts tree but, again, an empty design window.

But you're right -- a naked pod set has no appearance. But  not showing them is confusing people (in a way that not showing stages isn't), and they do have a location even before anything is added to them.  It seems like we need to do *something* to tell users they're there.

JoePfeiffer, 2022-07-11:
Oh, and whatever we do with showing pod sets, we need to do something similar with side boosters.  I've got a hunch the only reason we haven't had the same complaint with them is people are using pod sets a lot more.

neilweinstock, 2022-07-11:
Probably right.

hcraigmiller, 2022-07-12:
After considering everyone's comments, here's my view. . . 

As to pod markers, I like the suggestion for a small sphere (at the intersection of the radial distance, angle, and position) that does not change in size as the display is zoomed in and out; however, I really don't have an opinion as to whether crosshairs of some type should also be added.

But, as with the CG and CP markers, where users (including myself) use both the 2D and 3D views for presentations, the pod markers and any other similar markers need to be able to be turned on and off; I don't want phantom elements of any type appearing in presentation screen grabs.

neilweinstock, 2022-07-12:
I don't think that hiding the markings will be necessary if, as Joe suggests, they are only shown when the pod has no components attached.  I doubt anyone is going to leave unoccupied pods hanging around in their design.

jimmiedave, 2022-07-12:
Another thought might be the way the UI talks about it. It says "pods" and shows 3 round things on what I took for fins until now. Kinda "Mars Snooper"-like. Maybe if it was called "Pod Group" or "Add Pods", the function might be clearer. Not married to either of those phrases. Just something clear.

SiboVG, 2022-07-12:
Another addition to the visualization: I would make the color of the marking light grey, indicating that the marking is "optional" (can't think of a better word for it, but my meaning for it is that the marking does not represent a physical object; it's only to help users visualize it).

neilweinstock, 2022-07-12:
> Another addition to the visualization: I would make the color of the marking light grey, indicating that the marking is "optional" (can't think of a better word for it, but my meaning for it is that the marking does not represent a physical object; it's only to help users visualize it).

I like that.

Reminder to whomever implements this, eventually: make sure the objects are clickable.

SiboVG, 2022-07-12:
> Insofar as the marker turning off when an element is added to the pod, I would disagree with doing so. When working with pods and attached elements, it's nice to know where the reference point is actually located.

Agreed, but in that sense, having a sphere that is sized according to the radial distance etc. would be too visually obtrusive IMO. If you really would want to have a visual on the radial distance, maybe a proposition would be to add small horizontal lines that move according to the radial distance? (kinda reminds me of the [targeting computer from Star Wars](https://www.youtube.com/watch?v=cllbuMwLHm0))

[image]

Not that I'm too attached to the idea, but I also think that a plus sign is more subtle than a circle/sphere.

neilweinstock, 2022-07-12:
> > Insofar as the marker turning off when an element is added to the pod, I would disagree with doing so. When working with pods and attached elements, it's nice to know where the reference point is actually located.

How about it the pods only show whenever they're selected, or being edited?  I think that would satisfy my own needs.  Obviously that would eliminate the ability to click on them, but I think it would be perfectly fine to need to click them in the component tree to make them appear.

The more I think about it, the more I like this as a solution.  This means you see them as you're editing them, and then they disappear.

> Agreed, but in that sense, having a sphere that is sized according to the radial distance etc. would be too visually obtrusive IMO. If you really would want to have a visual on the radial distance, [...]

I feel rather strongly that they should always be displayed at the same size in the display, period.  And much smaller than your mockup there, something more like this:
[image]

Of course, while they're selected/being edited they would actually be highlighted with a colored outline.  Actually, if they *only* show while being edited, then they don't need to be grey in the first place.  They can be prominent while "active".

JoePfeiffer, 2022-07-12:
My intuition is it'll work best if they show whenever they don't have children in the tree. People who like to select in the design window seem pretty set on that (based on what I see on TRF).
After looking at the two mockups above, the sphere seems to give the idea of a "point you're hanging stuff on" beter.  Also, I'm no longer sure it can't be the same color as all the other parts in the design window.

jimmiedave, 2022-07-12:
A few quick-and-dirty ideas on other ways to communicate what this does:

[image]

neilweinstock, 2022-07-12:
> My intuition is it'll work best if they show whenever they don't have children in the tree. People who like to select in the design window seem pretty set on that (based on what I see on TRF).

Keep in mind that, if you only show them when they have no children, then they'll usually be visible only for a brief moment when the pods are first created, and then child components will be added and they will disappear.  So click-selecting these things is likely to be very rare.

neilweinstock, 2022-07-12:
> A few quick-and-dirty ideas on other ways to communicate what this does:

Aesthetically, I kind of like the look of the second one, although I'm not sure how it (or any of them) better convey the abstract concept of what exactly a pod set is.

Certainly we could change the name back to "Pod Set" if everyone things it's better (again, not sure how it's materially different, but I'll go along with whatever.)

hcraigmiller, 2022-07-12:
> A few quick-and-dirty ideas 

Like @neilweinstock, I prefer the second icon. I also prefer just "Pods" as you may have one, or maybe more, and the term "set" tends to indicate more than one.

jimmiedave, 2022-07-12:
To be clear, if the 4 were numbered 1, 2, 3, 4, are we talking about number 2 (where number 1 is the original)?

I'm fine with "Pods" if that's what the group thinks.

If you'd like me to tighten up the artwork, I can do so. If you let me know the specs, or the resource file/folder where I can find the original, I can get it happening.

neilweinstock, 2022-07-12:
Again, I’m not sure design two conveys the abstract concept of pods any better than what’s there now.

neilweinstock, 2022-07-12:
One more possibility: what if the pod markers are shown when *either* there are no children, *or* the set is currently selected? 
 This would mean that they're shown whenever it would be useful to see them, I think.

BTW not sure how to distinguish pods from boosters.  Whatever it is is likely to be somewhat arbitrary, but what can you do.  We could put a little "P" or "B" in the circle...
[image]

It's not beautiful but it's unambiguous.

hcraigmiller, 2022-07-12:
> One more possibility: what if the pod markers are shown when either there are no children, or the set is currently selected?
This would mean that they're shown whenever it would be useful to see them, I think.

That functionality works for me. Although just showing the markers when currently selected works for me too, maybe users should be reminded by a visual marker that an empty pod exists (use it or/to lose it, so to speak). 

As to the iconography, I still think 2 is more representative of what a pod is.

As to the style of the markers, I haven't seen anything yet that really strikes my fancy. . . I like the smaller size with identifier proposed by @neilweinstock, but I also like the precision offered by a smaller crosshair. . .

neilweinstock, 2022-07-12:
How is #3 better than what's there now?  Is it just the dotted lines?  I'm not sure what that even means.  It's not bad, I just don't see how it's an improvement.

> As to the style of the markers, I haven't seen anything yet that really strikes my fancy. . . I like the smaller size with identifier proposed by @neilweinstock, but I also like the precision offered by a smaller crosshair. . .

Just experimenting... what about something like this:
[image]

hcraigmiller, 2022-07-12:
> Just experimenting... what about something like this

Best, so far. . .

jimmiedave, 2022-07-12:
What I had in mind was that dashing the lines would show you're not making a physical structure, but a virtual one, a hierarchy, since I initially thought Pods would make a physical structure.

In my day job (I work in UX, and no, that doesn't make me "right" about anything at all), I remind people all the time that icon design is really hard, even for people who specialize in that. It's easy to design meaning for people who already know the meaning. It's really hard to design meaning for people who have never seen the icon before.

So this/these might have no better meaning for a set of fresh eyes. If you like, I can do an informal survey of my NAR section, find people who've never used Pods and see what they come up with.

DC

neilweinstock, 2022-07-12:
The dotted lines don't really speak to me, but if others think they're an improvement I have no objection.

jimmiedave, 2022-07-12:
I agree with @hcraigmiller on the representation in the drawing - @neilweinstock 's latest version is the best so far. It looks very much like it fits the "OpenRocket style"

JoePfeiffer, 2022-07-12:
> The dotted lines don't really speak to me, but if others think they're an improvement I have no objection.

To me they somehow convey that "virtualness".  A tiny change, but one I like.

> And yes, icon design is very hard. And ours our quite small.

I wonder if they're big enough for the dashes to even be visible?

neilweinstock, 2022-07-12:
> I wonder if they're big enough for the dashes to even be visible?

Probably not very clearly, especially on the "small" version.  But @jimmiedave will provide the answer. :)

PR #1535 (2022-07-18): [#1146] Add 2D visualization for pod sets
This PR fixes #1146. Podsets are now visualized by a marker (center square with lines to the north, east, south and west of the square) on the center of the podset and on every podset instance location. Instance marker have a darker grey color, while the podset center marker is slightly lighter to have a distinction between the two. The markers always remain visible, also when the podset has children. (**question:** should we add a checkbox option in the podset's configuration dialog to show/hide the markers? Or add a checkbox in the application prefer [... cut]
closed 2022-07-18 by JoePfeiffer