openrocket/openrocket #1150: Reading file with pods generates warning

neilweinstock, 2022-02-22:
This just cropped up.  Create a rocket with pods, save, and then open again.  I get this warning: 
[image]

Here's a file that demonstrates it.
[pods demo rocket.ork.zip](https://github.com/openrocket/openrocket/files/8113356/pods.demo.rocket.ork.zip)

JoePfeiffer, 2022-02-22:
After a brief look, I think this is slightly bigger and deeper than it appears (but should still be quick). It looks like we're using both angleoffset and radialdireciton to mean the same thing in different components.  We (fortunately) got bitten because pods are saving the parameter under the name radialdirection, but expecting to read it as angleoffset.  We need to adjust so they're consistent; personally, I think angleoffset seems to convey the idea more clearly.

An amusing line of code from RocketComponentSaver that illustrates why I think angleoffset is clearer:
elements.add("<radialdirection>" + angleOffset + "</radialdirection>");

SiboVG, 2022-02-22:
This is the result of the backward compatibility from PR #1110, right?

neilweinstock, 2022-02-22:
I assumed it was, but given that pods are new, they shouldn't be involved in any backwards compatibility hacks.  At least not intentionally. :)

SiboVG, 2022-02-22:
> I assumed it was, but given that pods are new, they shouldn't be involved in any backwards compatibility hacks. At least not intentionally. :)

Yeah, but the current code is to add 'radialdirection' to every AnglePositionable component (which a PodSet is), except for FinSet, TubeFinSet and RailButton:

```Java
if( c instanceof AnglePositionable ) {
	final AnglePositionable anglePos = (AnglePositionable)c;
	// The type names are currently equivalent to the enum names except for case.
	final String angleMethod = anglePos.getAngleMethod().name().toLowerCase(Locale.ENGLISH);
	final double angleOffset = anglePos.getAngleOffset()*180.0/Math.PI;
	elements.add("<angleoffset method=\"" + angleMethod + "\">" + angleOffset + "</angleoffset>");
	// TODO: delete this when no backward compatibility with OR 15.03 is needed anymore
	if (c instanceof FinSet || c instanceof TubeFinSet) {
		elements.add("<rotation>" + angleOffset + "</rotation>");
	}
	else if (!(c instanceof RailButton)) {
		elements.add("<radialdirection>" + angleOffset + "</radialdirection>");
	}
}
```

So changing `else if (!(c instanceof RailButton))` to `else if (!((c instanceof RailButton) || (c instanceof PodSet)))` already does the trick. (just did a quick test and it worked). Unless @JoePfeiffer thinks the issues is more deeply rooted?

closed 2022-02-22 by SiboVG

JoePfeiffer, 2022-02-22:
See #1152 -- it is more deeply rooted, but fixing the underlying problem isn't a priority.

PR #1151 (2022-02-22): Don't need to save <radialdirection> for pod sets, since there's no backward compatibility issue with them
closes #1150