openrocket/openrocket #1156: Linux 3D isn't working

JoePfeiffer, 2022-02-24:
Hadn't been verifying 3D was still working for Linux after my system rebuild last week; discovered tonight it's not.

When going to a 3D view I'm getting an error in the window: "Unable to load 3d Libraries: XXX0 profile[2]: GL2->profileImpl GL3bc !!! not mapped

Hoping distributing as an appimage will let me include old enough openGL libraries to get it working again....

SiboVG, 2022-02-24:
Well, well, well, looks like I'll be having my virtual beer from PR #1155 after all 😎

JoePfeiffer, 2022-02-24:
All too likely at the moment.

SiboVG, 2022-02-24:
98% sure it won't make a difference, but can you check if [this jar](https://drive.google.com/file/d/1WbkM2vrf-3LkcDlrHlMp5qIkxIP4llY-/view?usp=sharing) has the same issue? It uses a newer version of JOGL.

JoePfeiffer, 2022-02-24:
No change

SiboVG, 2022-02-24:
Maybe [useful forum question](https://jogamp.org/bugzilla/show_bug.cgi?id=1426) (although we don't use GLProfile.getMaxProgrammable or GLProfile.getMaxProgrammableCore).

FYI: I used JOGL v2.4.0-rc-20210111 in the JAR file I posted, downloaded [from here](https://jogamp.org/deployment/v2.4.0-rc-20210111/jar/).

SiboVG, 2022-02-24:
Aha @JoePfeiffer I think [this JAR](https://drive.google.com/file/d/1WbkM2vrf-3LkcDlrHlMp5qIkxIP4llY-/view?usp=sharing) fixes it (at least it fixed it on my machine). OR previously used JOGL packages for linux-amd64 and linux-i586. I also added packages for linux-aarch64 and linux-armv6hf.

JoePfeiffer, 2022-02-24:
Doesn't fix it on mine, rats.  Same error.  My machine is AMD64.  I'm not sure what the error message even means...

SiboVG, 2022-02-24:
Hm, crap... I'm running ARM64, so that explains why it now works for me with the added packages...

SiboVG, 2022-02-24:
Did you test running OR in different Java versions?

SiboVG, 2022-02-24:
I'm sorry @JoePfeiffer you'll have to be my tester for now, does [this JAR](https://drive.google.com/file/d/13uXXQjkZIfC9wv9O5rcIQUbRjA4NwcuM/view?usp=sharing) do anything?

JoePfeiffer, 2022-02-24:
You got it! Hmmm, now I've got to find a way to get you a real beer...

JoePfeiffer, 2022-02-24:
Are there MacOS arm jogl's we could try?  That could be the solution to that problem, too.

SiboVG, 2022-02-24:
> You got it! Hmmm, now I've got to find a way to get you a real beer...

Don't bother sending any American beers here, remember, I'm Belgian 😉

PR #1157 (2022-02-24): [fixes #1156] Update JOGL
This PR fixes #1156 by updating JOGL to v2.4.0-rc-20210111 + changing jogl-all.jar with a patched version taken from [here](https://www.astropixelprocessor.com/community/linux/app-1-077-not-run-on-kubuntu-18-04-lts/paged/3/#post-10955).
commit 08ec15d2 (2022-02-24): Merge pull request #1157 from SiboVG/issue-1156

[fixes #1156] Update JOGL
commit 21fa9c64 (2022-02-24): [fixes #1156] Update JOGL library + add more Linux distributions
commit f4d46bd9 (2022-02-24): [fixes #1156] Fix jogl-all
closed 2022-02-24 by SiboVG