openrocket/openrocket #1160: Update Mac installer to single bundle with Applications folder target

neilweinstock, 2022-02-25:
Something like this:
[image]

neilweinstock, 2022-02-26:
This seems rather well-explained here: https://www.ej-technologies.com/resources/install4j/v/8.0/help/doc/concepts/dmgStyling.html

I am not convinced that, at least initially, we need the custom background, just the Applications shortcut.  Although a background that includes that Arrow would be nice, I suppose.  Maybe I'll try to concoct one.

neilweinstock, 2022-02-26:
Maybe you can try this as a background image for the .dmg window.
[image]

SiboVG, 2022-02-27:
Custom installation backdrops always bring a small smile on my face, so I'm definitely pro. Maybe my one concern will be that the black backdrop will cause bad contrast between the background and application icon/folder and text.

neilweinstock, 2022-02-27:
Argh, you might be right.  On the desktop, the OS automatically chooses a font color to contrast with the background... but in a finder window, seemingly, it does not.  WTF.

So with the above background, you won't see the names until you select one of them, then it shows up in white.

What other background would make sense?  The arrow is useful if nothing else, but I was hoping to do something fun in the background.  If it's going to be too much of a problem, then I'll bag the idea.  In the meantime, here's an image with just the arrow.

[image]

SiboVG, 2022-02-27:
Maybe putting semi-transparent (e.g. 10% opacity) white rectangles with feathered edges in the spots where the 2 items will be can help.

neilweinstock, 2022-02-27:
That'll work if I know exactly where they'll be.  I'd first need an installer with the Applications folder in place, then I could create the image, and then we could apply it back into the installer.  That's a bit of effort; for the first go-round I guess let's just use the arrow, and I can add the stars later.

SiboVG, 2022-02-27:
Also I'd propose to use the starfield from OR's website, it's a bit cleaner and it's a nice touch to use the same image.

neilweinstock, 2022-02-28:
Ok, here is is with the starfield from openrocket.info.  I don't know if we'll be able to use, but just in case...
[image]

commit 196c5125 (2022-03-07): Merge pull request #1203 from SiboVG/issue-1160

[fixes #1160] Update Mac installer to single bundle with Applications folder target
commit 8e75c577 (2022-03-07): [fixes #1160] Update Mac installer to single bundle with Applications folder target
closed 2022-03-07 by JoePfeiffer