openrocket/openrocket #1163: Doesn't create new simulation from new configuration reliably.

JoePfeiffer, 2022-02-26:
1) Create a new design, with no motor configurations nor simulations
2) Create a new motor configuration, including selecting a motor
3) Switch to the simulation tab. A simulation with no motors appears as Simulation 1. If you create a new simulation, it will have the motor selected in the previous configuration..

JoePfeiffer, 2022-02-26:
Another possibly related issue.
[image]
ork file:
[Composite-Warehouse-Air-Strike-3829.zip](https://github.com/openrocket/openrocket/files/8147296/Composite-Warehouse-Air-Strike-3829.zip)

neilweinstock, 2022-02-26:
Man, I have never experienced any of these, just checked and definitely do not see this.  Are there screenshots to show what's going on?

JoePfeiffer, 2022-02-26:
The sequence I described in the initial bug report does it quite reliably for me.  I haven't looked at the others to see if I can replicate them.  I think the guy on FB also reported it in another group and had screenshots; I'll look over there again.

JoePfeiffer, 2022-02-26:
Edited my previous comment with user screenshot.

neilweinstock, 2022-02-26:
Huh.  Is that just because the window is too short?  I am baffled by that screenshot.

I am definitely seeing, in Brian's sample file, that the previously-configured CTI motors are not being recognized by the beta.  Is that connected to the previous changes in how motors are identified in the program?

JoePfeiffer, 2022-02-26:
Yes.  I haven't filed that issue yet, but a few motors in old simulations aren't being correctly identified -- one of the things I need to seriously get on.

neilweinstock, 2022-02-26:
Filed as #1174.

SiboVG, 2022-02-27:
> Huh. Is that just because the window is too short? I am baffled by that screenshot.

I have a feeling that this is the issue... @JoePfeiffer any way to confirm this with Brian?

JoePfeiffer, 2022-02-27:
Rereading this, I think he's reporting the same problem in two places -- he's seeing a single line in the motor manufacturer table in the motor selection, and also in selecting a configuration.  If I deliberately resize my window so it's too short, I can duplicate both of them.

Ah, then if I exit and restart OR, the window comes up the same size as when I closed it previously, and has the too-short tables.  I think that's it, also it's not related to the other errors we're seeing reported here.

I've asked him to try expanding his window vertically to see what happens.

JoePfeiffer, 2022-02-27:
Finally got a couple of good screenshots from him.  First is what he sees when he first starts OR and goes to the Motors&Configurations tab
[image]
[image]
[image]

SiboVG, 2022-02-27:
Yeah that's all "normal" behavior. The user just has a small resolution screen which causes the top part of the window to be more squashed together.

Anyway, I made a change that adds a minimum height to the 2 places that the user mentioned. Will post it in the PR that fixes the rest of this issue.

neilweinstock, 2022-02-27:
It does seem like there is quite a bit of excess padding there., both at the bottom of the top half and the top of the bottom half.  Perhaps that can be trimmed a bit?

Most users will be using a higher-res screen than that, but I still think we could be a bit more efficient.

SiboVG, 2022-02-27:
> It does seem like there is quite a bit of excess padding there., both at the bottom of the top half and the top of the bottom half. Perhaps that can be trimmed a bit?

Good point, I'll play with it a bit.

JoePfeiffer, 2022-02-27:
Is it possible to define the layout in such a way as to use the space when screen resolution is available (on my screen it looks quite nice), and reduce the space between windows to a minimum before reducing the size of windows when it's not?

SiboVG, 2022-02-27:
> Is it possible to define the layout in such a way as to use the space when screen resolution is available (on my screen it looks quite nice), and reduce the space between windows to a minimum before reducing the size of windows when it's not?

Hm that gets a bit tricky if you still want other components to also scale when resizing a window; Java Swing layouts are not terribly flexible in that regard (or at least in my experience, I know there are some tricks like setting a growth order, but that also often doesn't work the way that you want...)

JoePfeiffer, 2022-02-28:
I guess I should have said "reasonable", not "possible." I can see a little bit of vertical padding we can cut, but I'm not sure how far we should go in trying to support antique hardware.

SiboVG, 2022-02-28:
> I guess I should have said "reasonable", not "possible." I can see a little bit of vertical padding we can cut, but I'm not sure how far we should go in trying to support antique hardware.

Yeah I already cut come vertical padding which saved some space, but I would indeed not spend too much time on it... The addition of the minimum height to the components also helps alleviate the issue, so I guess we're covered enough

neilweinstock, 2022-02-28:
> Yeah I already cut come vertical padding which saved some space, but I would indeed not spend too much time on it... The addition of the minimum height to the components also helps alleviate the issue, so I guess we're covered enough

Agreed.  I would keep in mind, though, that this issue with screen space utilization is not in fact related to the original issue reported up top, so let's not close this one yet (or we can split it off into another, or whatever).

commit 7e14a07f (2022-02-28): Merge pull request #1201 from SiboVG/issues-1163

[fixes #1163 & #1175] Fix unreliable simulation creation from new flight config
commit cd44d73b (2022-02-28): [fixes #1163] I messed up one line of code...
closed 2022-02-28 by SiboVG