openrocket/openrocket #1164: Sign flip in fin component offset when opening .rkt file

jimmiedave, 2022-02-26:
Opening "Estes Argent.rkt", things generally look good, but the fins are in the middle of the body. I initially wondered whether this is due to the model having two coupled tubes in the main body. Dunno.

[image]

However, intuition said that flipping the sign on the fin offset would put the fins where they belong, and this appears to work.

[image]

Not sure whether this occurs with other components, but a quick check shows the sign-flip on fins in other .rkt files.

hcraigmiller, 2022-02-26:
BUG CONFIRMED: On import of ,rkt file, **the pos/neg sign of fin reference point** is reversed when referenced from the _tip of the nose cone_, and when referenced from the _top of the parent component_. When referenced from the _bottom of the parent component_, the pos/neg sign is correct.

SiboVG, 2022-02-27:
Does exporting an OpenRocket-design to Rocksim also have the same fin flip issue inside of Rocksim?

hcraigmiller, 2022-02-28:
Upon export, Rocksim opens with the fins correctly placed, in reference to all points.

PR #1186 (2022-02-28): [fixes #1164] Account for different RockSim FinSet location directions
This PR fixes #1164 where the location of FinSets from a Rocksim import would be inverted for locations that were referenced from the tip of the nose cone and the top of the parent component.

Here is a [jar file](https://github.com/openrocket/openrocket/suites/5467350835/artifacts/174098659) for testing.
commit 47526e32 (2022-02-28): Merge pull request #1186 from SiboVG/issue-1164

[fixes #1164] Account for different RockSim FinSet location directions
commit d2acd0ed (2022-02-28): [fixes #1164] Account for different RockSim FinSet location directions
closed 2022-02-28 by JoePfeiffer