openrocket/openrocket #1165: Error On Launching the 22.02.beta.01 JAR file

silversurfer77, 2022-02-26:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Downloaded the new/beta 22.02.beta.01 .JAR file (not the installer)
2. Downloaded and Installed Java 64 bit

- java version "1.8.0_321"
- Java(TM) SE Runtime Environment (build 1.8.0_321-b07)
- Java HotSpot(TM) 64-Bit Server VM (build 25.321-b07, mixed mode)

4. Running on Windows 11
5. Double clicked the .JAR file and OpenRocket opened and an error dialog appeared immediately
6. I click CLOSE on the error dialog box and OR seems to be working OK...but I haven't really tested anything yet.
Include your email address (optional; it helps if we can contact you in case we need additional information):
dankunkel@gmail.com

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NoSuchMethodError: java.util.List.of(Ljava/lang/Object;)Ljava/util/List; at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.getLatestReleaseJSON(UpdateInfoRetriever.java:362) at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.runUpdateFetcher(UpdateInfoRetriever.java:142) at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.run(UpdateInfoRetriever.java:123)

---------- Thread information ----------
Thread[UpdateInfoFetcher,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.01
OpenRocket source: default
OpenRocket location: C:\Users\danku\OneDrive\Desktop\OpenRocket-22.02.beta.01.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
file.encoding=Cp1252
file.encoding.pkg=sun.io
file.separator=\
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Users\danku\OneDrive\Desktop\OpenRocket-22.02.beta.01.jar
java.class.version=52.0
java.endorsed.dirs=C:\Program Files\Java\jre1.8.0_321\lib\endorsed
java.ext.dirs=C:\Program Files\Java\jre1.8.0_321\lib\ext;C:\windows\Sun\Java\lib\ext
java.home=C:\Program Files\Java\jre1.8.0_321
java.io.tmpdir=C:\Users\danku\AppData\Local\Temp\
java.library.path=C:\Program Files\Java\jre1.8.0_321\bin;C:\windows\Sun\Java\bin;C:\windows\system32;C:\windows;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Users\danku\AppData\Local\Microsoft\WindowsApps;;.
java.runtime.name=Java(TM) SE Runtime Environment
java.runtime.version=1.8.0_321-b07
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=1.8
java.util.Arrays.useLegacyMergeSort=true
java.vendor=Oracle Corporation
java.vendor.url=http://java.oracle.com/
java.vendor.url.bug=http://bugreport.sun.com/bugreport/
java.version=1.8.0_321
java.vm.info=mixed mode
java.vm.name=Java HotSpot(TM) 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=1.8
java.vm.vendor=Oracle Corporation
java.vm.version=25.321-b07
jnlp.jogamp.tmp.cache.root=jln2663370734863557396
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 11
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
[... 127 lines cut ...]
151 3.752 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
152 3.752 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
153 3.752 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
154 3.768 DEBUG (GUIUtil.java:311) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1936,height=1048]
155 3.768 DEBUG (GUIUtil.java:316) Storing maximized state of net.sf.openrocket.gui.main.BasicFrame
156 3.768 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
157 3.768 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
158 3.768 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
159 3.768 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
160 3.788 DEBUG (GUIUtil.java:311) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1936,height=1048]
161 3.788 DEBUG (GUIUtil.java:316) Storing maximized state of net.sf.openrocket.gui.main.BasicFrame
162 3.788 DEBUG (UpdateInfoRetriever.java:229) Update checker response code: 200
163 3.817 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
164 3.817 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.468577 user scale, 1771.00 overall scale
165 3.964 ERROR (SwingExceptionHandler.java:48) Handling uncaught exception on thread=Thread[UpdateInfoFetcher,6,main]
> java.lang.NoSuchMethodError: java.util.List.of(Ljava/lang/Object;)Ljava/util/List; > at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.getLatestReleaseJSON(UpdateInfoRetriever.java:362) > at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.runUpdateFetcher(UpdateInfoRetriever.java:142) > at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.run(UpdateInfoRetriever.java:123)
166 3.964 STDERR (SwingExceptionHandler.java:49) java.lang.NoSuchMethodError: java.util.List.of(Ljava/lang/Object;)Ljava/util/List;
167 3.964 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.getLatestReleaseJSON(UpdateInfoRetriever.java:362)
168 3.964 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.runUpdateFetcher(UpdateInfoRetriever.java:142)
169 3.964 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.communication.UpdateInfoRetriever$UpdateInfoFetcher.run(UpdateInfoRetriever.java:123)
170 3.964 INFO (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
171 3.964 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
172 19.516 USER (SwingExceptionHandler.java:220) User requested sending bug report
173 19.518 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2022-02-26:
The new release will not run on Java 8 (getting away from that prehistoric version was one of the goals!).  If you must run the jar for some reason, you need to be using at least (and preferably) Java 11. The packaged installers include a copy of the Java 11 runtime.

hcraigmiller, 2022-02-26:
BUG CONFIRMED: On first run of the .jar file, the following message appeared. After clicking _Okay_, the message disappeared and OR opened with no detected anomalous behavior. Closing and reopening OR did not trigger a repeat of the message.

[image]

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

JoePfeiffer, 2022-02-26:
Now I'm confused -- I didn't think OR would run at all under Java 8 (and my first response was too hasty).

neilweinstock, 2022-02-26:
My recollection from the past is that it kind of runs, sometimes, with some problems.  Definitely not recommended.  Is there a way to check for that in the app and refuse to run (or put up a helpful message) if a bad JRE is used?

hcraigmiller, 2022-02-26:
Jar file warning added to Wiki at http://wiki.openrocket.info/Other_Installations, with instructions.

neilweinstock, 2022-02-26:
Closing this because it's a documentation issue, and that is "fixed".

silversurfer77, 2022-02-27:
It appears that you cannot download and install the Java 11 JRE. So unless you are a developer with a license, the JAR is kinda useless. I prefer the JAR file since it is portable and doesn't need to be installed.

I am hoping it is possible to install OR using the installer, then uninstall, and preserve Java 11 so I can run the standalone JAR.

SiboVG, 2022-02-28:
> It appears that you cannot download and install the Java 11 JRE. So unless you are a developer with a license, the JAR is kinda useless. I prefer the JAR file since it is portable and doesn't need to be installed.

There are tools to switch between Java-versions. jEnv is a very good one for Linux and macOS. I believe there are similar alternatives for Windows. If you're really having troubles with installing Java 11 JRE I'd suggest you check out a tool like jEnv.

closed 2022-02-28 by neilweinstock

JoePfeiffer, 2022-02-28:
> It appears that you cannot download and install the Java 11 JRE. So unless you are a developer with a license, the JAR is kinda useless. I prefer the JAR file since it is portable and doesn't need [...]

Note JRE is, to the best of my knowledge, not available separately whether you've got a license or not.  You can certainly download the JDK -- I got it for my Win10 box from https://adoptium.net/releases.html?variant=openjdk11, it works fine, and I have no license.