openrocket/openrocket #1166: In the rocket view, motor configuration dropdown box doesn't have the motor names

CarVac, 2022-02-26:
[image]

Left is 15.03, right is the new 22.02 beta 1.

On Pop!OS Linux, running the .jar file using OpenJDK 11.0.13 2021-10-19 (yes I know this isn't the recommended configuration, but I prefer a standalone file like the AppImage and there isn't an AppImage available).

JoePfeiffer, 2022-02-26:
I'm not seeing this on my Debian box -- could you send me (or add it to this bug report) your .ork file? You'll need to zip it first for github to accept the upload.

hcraigmiller, 2022-02-26:
Could not reproduce.

[image]

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

CarVac, 2022-02-26:
[Area_Rule_3.0.zip](https://github.com/openrocket/openrocket/files/8146563/Area_Rule_3.0.zip)

I have a nickname on the configurations that tell me what case the motor goes with. I guess the syntax for including the currently-selected motor isn't respected anymore?

In 15.03, when I create a new configuration and try to rename it it defaults to `[{motors}]` which I just prefixed by the case and manufacturer, but on 22.02 it defaults to the actual name of the currently selected motor.

JoePfeiffer, 2022-02-26:
Inadvertently.  This is a regression we need to look at.

commit 804d9bad (2022-02-28): Merge pull request #1180 from SiboVG/issue-1166

[fixes #1166] Use [{motors}] for flight config motors placeholder
commit d3ec4f66 (2022-02-28): [fixes #1166] Use [{motors}] for flight config motors placeholder
closed 2022-02-28 by JoePfeiffer