openrocket/openrocket #1168: [New feature request] Option to turn CG and CP markings on and off in the 3D views

hcraigmiller, 2022-02-26:
In the current unstable, in the _3D views_, the visible CG and CP markings interfere with the user's ability to take screen shots for esthetic purposes, particularly in the _3D Finished_ view. TRF user K'Tesh has asked about an option to allow these markings to be turned on and off.

SiboVG, 2022-02-27:
Would also be a good way to mask the CG/CP issues in the 3D view on Mac's 🙃

RemingtonHolder, 2022-03-25:
Just tinkering around, this is a surprisingly frustrating thing to do. The code for CG and CP points seems to be pretty deeply integrated to just always exist. 🤷

SiboVG, 2022-03-25:
> Just tinkering around, this is a surprisingly frustrating thing to do. The code for CG and CP points seems to be pretty deeply integrated to just always exist/display. 🤷

It's not that hard to do in fact, if you comment out `drawCarets(gl, glu);` on `line 327` in `swing/src/net/sf/openrocket/gui/figure3d/RocketFigure3d.java`, the markings are gone:
[image]

So the only thing that would need to be added for this PR is create a checkbox in the preferences dialog, store that checkbox value in preferences and create a simple if-condition on `line 327` that if the checkbox is checked, that you should call `drawCarets(gl, glu);`.

RemingtonHolder, 2022-03-25:
Seems as if I were looking at the wrong things. 😂

RemingtonHolder, 2022-03-26:
@SiboVG Do you think a preference for this would best fit in the Design or Graphics preference tab.

JoePfeiffer, 2022-03-26:
My first thought is that it could be a button in the rocket figure toolbar -- there's plenty of room for it there, and a user could turn it on/off without having to go to another tab or something.  We could disable it for the non-3D views (much like the scale buttons are disabled for 3D views now -- though I think there's a consensus we'd like that to change that some day).  For that matter, there's no compelling reason to not let users turn them off in the non-3D views as well.

hcraigmiller, 2022-05-31:
Fix complete by way of PR #1393.

closed 2022-05-31 by hcraigmiller