openrocket/openrocket #1171: Drag coefficient override has no effect on simulation results

neilweinstock, 2022-02-26:
Tested on macOS 12.

SiboVG, 2022-02-27:
I think the slider is just messed up. If you type in Cd values in the typebox, then it works fine, same goes for when you use extreme values in the slider. But In the first half of the slider, the sims don't get affected. It's just a weird slider, also looks like it is a logarithmic slider instead of a linear one.

https://user-images.githubusercontent.com/11031519/155882061-2628bb1a-8595-47d5-9b3f-85bf595daeb5.mov

Do we want to have this logarithmic behavior? I must say I was also confused the first time I used it; thought the slider just wasn't working at all until I got to the end of the slider.

neilweinstock, 2022-02-27:
I agree that the exponential slides is bad, and should be changed.  However, I cannot replicate your results where changing the value actually changes the simulation.  On my rocket, changing the setting from 0-1 makes no difference in the sim results.  That includes the simple model rocket example that you're showing above.

SiboVG, 2022-02-27:
That's odd, I can clearly see the values changing in the simulations tab...

neilweinstock, 2022-02-27:
I'm trying to think of what could possibly be different between our setups, and coming up empty so far.

SiboVG, 2022-02-27:
Could you maybe send a quick demo of what you're seeing in 'A simple model rocket'? (you can make a screen recording on Mac using 'Cmd + Shift + 5' and use 'Record Selected Portion' to only record OR's screen are. When you're done recording just go to hit the stop-button in the menu bar in the top right)

neilweinstock, 2022-02-27:
https://user-images.githubusercontent.com/243872/155891060-962be4b7-2006-4789-8a32-5fa2e256c664.mov

SiboVG, 2022-02-27:
Aha, I see what you did, you are overriding the Cd of the stage, not of an individual component like the nose cone. So I guess the problem is that Cd overriding for stages doesn't work.

neilweinstock, 2022-02-27:
Yes, that is what I assumed was the most likely use for the override.

SiboVG, 2022-02-27:
> Yes, that is what I assumed was the most likely use for the override.

Yes, most likely, just wasn't on the same page as you :)

PR #1627 (2022-09-02): Rewrite CD overrides so it works for component assemblies
Fixes #1171
closed 2022-09-02 by JoePfeiffer