openrocket/openrocket #1173: Putting fins on transitions shows less accurate aerodynamics than the old 0-length body tube hack.

JoePfeiffer, 2022-02-26:
A user on FB reports CP on a V2 model using the the 0-length body tube hack is very close for OR and RockSim.  Attaching fins to transition moves CP unreasonably far forward.

RockSim image:
[image]
RockSim file:
[V2_BT101rkt.zip](https://github.com/openrocket/openrocket/files/8147190/V2_BT101rkt.zip)

0-length body tube hack image:
[image]
OR file:
[V2_BT101ork.zip](https://github.com/openrocket/openrocket/files/8147193/V2_BT101ork.zip)

Fins on transition image:
[image]
OR file:
[V2_BT101_RS-22.zip](https://github.com/openrocket/openrocket/files/8147195/V2_BT101_RS-22.zip)

JoePfeiffer, 2022-03-02:
Got another detail from the original reporter last night:

> It dawned on me last night that OR is using straight Barrowman for CP calcs whereas I used RockSims' calculation for the RS
pic I sent.  I opened up the V2 in RS and changed to CP calc method, and it looks close to the OR 22 sim I sent you. 
"mischief managed"
[image]

I'm still troubled by it

hcraigmiller, 2022-03-18:
As to the original issue title of "Putting fins on transitions shows less accurate aerodynamics than the old 0-length body tube hack", a comparison between the current unstable, Rocksim 10, and RASAero II, using an airframe compatible with all three, shows very little difference in altitude in some cases, and a huge difference between OpenRocket and Rocksim and RASAero II in others [all simulations have polished surfaces using Barrowman].

[image]

Rocksim 10 and RASAero II seem to be fairly close to one another.

If this is helpful, I can do additional comparisons.

JoePfeiffer, 2022-03-19:
I'd be very curious to see how results compare using the same rocket and motors, but no boat tail (same diameter all the way back).

hcraigmiller, 2022-03-19:
Here are the results:

[image]

JoePfeiffer, 2022-03-20:
Looks like whatever is causing those wildly optimist results, it isn't the transition.  Could you post the .ork for the straight-tube version?

hcraigmiller, 2022-03-20:
Here are the files for the straight body... I removed the rail buttons so that all three are exactly the same.

[02.01.Comparison.Rev_02.ork.txt](https://github.com/openrocket/openrocket/files/8310437/02.01.Comparison.Rev_02.ork.txt)
[02.02.Comparison.Rev_02.rkt.txt](https://github.com/openrocket/openrocket/files/8310438/02.02.Comparison.Rev_02.rkt.txt)
[02.03.Comparison.Rev_02.CDX1.txt](https://github.com/openrocket/openrocket/files/8310439/02.03.Comparison.Rev_02.CDX1.txt)

PR #1247 (2022-03-20): Get correct fin span when last point in shape is lower than first point.
Note that the lowest point on the fin is guaranteed to be either the first point which is at (0, 0) or the last point, which will have a y coordinate of 0 if on a body tube, greater than 0 if on an increasing transition, or less than 0 on a decreasing transition (boattail).

Fixes #1173 and #1243

I'm leery of using either RockSim or a nasty hack as ground truth, but the CP is in a much more reasonable place using the .ork file from the report of #1173
[V2_BT101_RS-22.zip](https://github.com/openrocket/openrocket/files/82 [... cut]
closed 2022-03-20 by wolsen