openrocket/openrocket #1184: [Bug] Bug report using simulation and pods

SiboVG, 2022-02-27:
<details>
  <summary>Bug Report</summary>

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Running simulation with motors in booster pods

1.Motors in booster pods C6-0

2.Motor in main C6-5

3.

Include your email address (optional; it helps if we can contact you in case we need additional information):

sj_h1@live.com

(Do not modify anything below this line.)

---------- Exception stack trace ----------

net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120) at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:414) at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:98) at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847) at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857) at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317) at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233) at java.desktop/java.awt.Dialog.show(Dialog.java:1070) at java.desktop/java.awt.Component.show(Component.java:1716) at java.desktop/java.awt.Component.setVisible(Component.java:1663) at java.desktop/java.awt.Window.setVisible(Window.java:1031) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005) at net.sf.openrocket.gui.main.SimulationPanel$3.actionPerformed(SimulationPanel.java:177) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.lang.NullPointerException

---------- Thread information ----------

Thread[AWT-EventQueue-0,6,main]

---------- System information ----------

OpenRocket version: 22.02.beta.01

OpenRocket source: default

OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar

JOGL version: 2.4.0-rc-20210111

Current default locale: en_US

System properties:

awt.toolkit=sun.awt.windows.WToolkit

exe4j.consoleCodepage=cp0

exe4j.isInstall4j=true

exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe

exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe

exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0

exe4j.tempDir=

exe4j.unextractedPosition=0

file.encoding=Cp1252

file.separator=\

install4j.appDir=C:\Program Files\OpenRocket\

install4j.exeDir=C:\Program Files\OpenRocket\

install4j.launcherId=59

install4j.swt=false
[... 613 lines cut ...]
73386 3132.448 DEBUG (FlightData.java:247) Computed flight values: maxAltitude=396.9182820667803 maxVelocity=5.188603422566388E10 maxAcceleration=1.245550891024932E14 maxMachNumber=203638.53407852934 timeToApogee=2.791396636777916 flightTime=4.241396636777928 groundHitVelocity=NaN launchRodVelocity=23.067149710989064

73387 3132.448 INFO (BasicEventSimulationEngine.java:94) <<Finished simulating branch: Sustainer curTime:4.241396636777928 finTime:4.241396636777928

73388 3132.448 INFO (BasicEventSimulationEngine.java:88) >>Starting simulation of branch: Booster Set

73389 3132.448 VBOSE (BasicEventSimulationEngine.java:263) HandleEvents: current branch = Booster Set

73390 3132.448 VBOSE (BasicEventSimulationEngine.java:137) Taking simulation step at t=0.0 altitude 0.0

73391 3132.482 ERROR (SwingExceptionHandler.java:119) An exception occurred during the simulation

> java.lang.NullPointerException

73392 3132.482 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog

73393 3132.482 INFO (SwingExceptionHandler.java:201) Showing Exception dialog

73394 3135.317 USER (SwingExceptionHandler.java:220) User requested sending bug report

73395 3135.318 DEBUG (JarUtil.java:40) Found jar file using codeSource

---------- End of bug report ----------

</details>

SiboVG, 2022-02-28:
The NullPointerException is caused by `BarrowmanCalculator.java`, more specifically by the FinSetCalc being null because `FinSet f` is not present in `calcMap`:
```Java
for (RocketComponent c : configuration.getActiveComponents()) {
	if (c instanceof FinSet) {
		FinSet f = (FinSet) c;
		FinSetCalc finSetCalc = (FinSetCalc) calcMap.get(f);
		Coordinate coord = new Coordinate(finSetCalc.getMidchordPos());
		mul += 0.6 * Math.min(f.getFinCount(), 4) * f.getPlanformArea() *
				MathUtil.pow3(Math.abs(f.toAbsolute(coord)[0].x - cgx)) /
				(conditions.getRefArea() * conditions.getRefLength());
	}
}
```
(I rewrote the code a bit so that the NullPointerException became more clear)

I think the root cause is with the use of `getActiveComponents` inside both the `getDampingMultiplier` method (where the code snippet above is from) and in the `buildCalcMap` method, which populates `calcMap`. The `getActiveComponents` method is denoted as deprecated in the code with the comment `recommend migrating to either: 'getAllComponents' or 'getActiveInstances'`.

I must admit that I am not knowledgeable enough right now to make any changes on this code. Since it's an important piece of code (changing it will change simulation results), I'm not comfortable tackling this issue without the proper knowledge...

JoePfeiffer, 2022-03-01:
I'll give it a look

sjh114961, 2022-03-02:
@JoePfeiffer please change the status of this. It is obviously NOT closed.

SiboVG, 2022-03-02:
@sjh114961 this issue is still marked as open. #1195 was closed because it is a duplicate of this issue.

JoePfeiffer, 2022-03-03:
Ignition of core stage and parallel stages both at launch
[pod-example-2.zip](https://github.com/openrocket/openrocket/files/8174840/pod-example-2.zip)

PR #1216 (2022-03-06): Create CalcMap based on all components, not just active ones
Fixes #1184
closed 2022-03-06 by SiboVG