openrocket/openrocket #1187: Better color picker for Linux

neilweinstock, 2022-02-28:
I was just astonished to learn that in Linux, the color picker is this steaming pile of garbage:
[image]
Apart from its other glaring deficiencies, this picker does not apparently (at least as invoked by OR) support alpha channel, which kneecaps one of the features of the new release.

I am no fan of the Windows/Mac picker but at least it's functional.

Is there a way to get a more functional picker into the Linux builds?  Or at the very least, enable alpha channel operation?

Better long term solution for alpha is an opacity slider in the appearance panel, but until we get that the picker should offer it.

JoePfeiffer, 2022-02-28:
By default, Java applications are supposed to use a look and feel called "Metal", which as I understand it would make the application look similar across all platforms.  The Metal l&f supports transparency. OR actually has code looking for and forbidding the use of the Metal look and feel, in GUIUtil.java, on the grounds that it's ugly (and... it is). You can't even set it on the command line.

The GTK color chooser underlying the implementation in Linux actually does support a transparency slider. I haven't spent a long time on it, but I haven't found a way to get the Java JColorChooser to enable it (I've got no idea why that isn't the default).

neilweinstock, 2022-02-28:
Is the code actively disabling the RGB and Color Swatch panels?  Can't figure out why they're there in Mac & Win and not Linux.

This is of course a lower-priority item.

JoePfeiffer, 2022-02-28:
#1197 fixes the functional issue here.

closed 2022-02-28 by JoePfeiffer