openrocket/openrocket #1189: "Shine" slider in Linux Appearance tab has (apparently) zero width

jimmiedave, 2022-02-28:
[image]

Shine slider looks like a nubbin in a box on Linux. Works as expected on macOS.

SiboVG, 2022-02-28:
@jimmiedave since my machine doesn't suffer from the issue, can you verify that [this JAR](https://drive.google.com/file/d/1DtnsOs2Ih-M6W4DqHQlDTOZQnE_RWumY/view?usp=sharing) fixes the issue?

jimmiedave, 2022-02-28:
Running the JAR gives (only) this message in the terminal:

`Gtk-Message: 11:32:52.297: Failed to load module "canberra-gtk-module"`

Related?

SiboVG, 2022-02-28:
> Running the JAR gives (only) this message in the terminal:
> 
> `Gtk-Message: 11:32:52.297: Failed to load module "canberra-gtk-module"`
> 
> Related?

No, I also get that message, but the slider still works on my Ubuntu build. (I also haven't researched what the Gtk-Message means, I chose to peacefully ignore it)

SiboVG, 2022-02-28:
> Reboot of machine doesn't improve matters.
> 
> Could the 10x value thing be related? Maybe not enough pixels to encompass a range of 0-1000 making the widget be weird?

Hm, I don't know, but it's just very strange that the slider button also differs from the rotation slider button. Before this PR I had the same, round button as in the last screenshot you posted, but after the JAR, on my machine the slider button changed to the same one as the rotation slider...

jimmiedave, 2022-02-28:
Uninstalled (purged, with aptitude, actually) Ubuntu's default-jre and default-jre-headless, reinstalled, noted same results.

Also noted in fiddling with this that using the arrow gadgets on the Shine value box (i.e. not the slider) will allow negative values. Should probably be clipped to 0%. Do you want a fresh bug on this?

SiboVG, 2022-02-28:
By the way you can get rid of the Gtk-Message by running `sudo apt install libcanberra-gtk-module`

SiboVG, 2022-02-28:
Myeah something must be wrong... And I'll include the shine value clipping in the PR of this issue, no need to make it a separate issue.

SiboVG, 2022-02-28:
Okay just so I'm certain, does [this JAR](https://drive.google.com/file/d/134MyTskKzKVDzEY2eTRBTjfdVOSsniPT/view?usp=sharing) change anything at all? (this JAR btw also has the clamping so you can play around with that)

jimmiedave, 2022-02-28:
Yes, correct slider, and the clamp works great!

Only thing left: value window changes sizes, making the slider a little jittery as you drag it, but that's so minor, I think it's livable:

[image]
[image]

SiboVG, 2022-02-28:
> Only thing left: value window changes sizes, making the slider a little jittery as you drag it, but that's so minor, I think it's livable:

Let's file that as a problem for future me 😉

jimmiedave, 2022-02-28:
Sounds good - the Rotation slider on the same panel has the same jitter thing, and different sliders in panels in the material tabset deal with this in different ways - so a little bigger than fixing the single jitter.

Appreciate your quick work on this.

PR #1199 (2022-02-28): [fixes #1189] Fix Shine slider too small
This PR fixes #1189 where the shine slider was too small because it used a JSlider instead of BasicSlider.

Additionally, this PR also adds clamping for both the opacity and shine slider so that it is limited to 0 - 100 %.

(Extra: apparently the changes from PR #970 were not present in unstable, no clue why because normally it got merged but anyway, now it's back)

(Already tested in the #1189 thread).
commit 6fe0a2bf (2022-02-28): Merge pull request #1199 from SiboVG/issue-1189

[fixes #1189] Fix Shine slider too small
commit fb872fdb (2022-02-28): [fixes #1189] Fix Shine slider zero width
closed 2022-02-28 by SiboVG