openrocket/openrocket #1192: Request: default component color something other than black

jimmiedave, 2022-02-28:
Suggest we make default color something like #7F7F7F so that the black and white boxes in the preview area initially make some intuitive sense to users. They've never made sense to me until just a moment ago when I opened one of the sample files that was colored up (something I've never really used OR for).

The black-and-white preview boxes aren't labeled (other than being in the preview area), and unless the color chosen is not black (which is one of the colors in the boxes that's being used for preview against your chosen colors), they just look like a bunch of weird black and white boxes, and it's not clear that you can see your text on the sample backgrounds (because black-text-on-black and black-text-on-background-and-white don't give you the sense that you're looking at your chosen color).

neilweinstock, 2022-02-28:
This is an interesting thought.

In general, I think I would like to see all the default colors be the same, and a mid-gray would make sense for that.

SiboVG, 2022-02-28:
So you want something like this?
[image]

neilweinstock, 2022-02-28:
Hmm, that's a bit dark... I propose BBBBBB

SiboVG, 2022-02-28:
Fine by me, here's the #BBBBBB result:
[image]

neilweinstock, 2022-02-28:
Works for me.  I'm sure people will complain. :)

jimmiedave, 2022-02-28:
I'm fine with the #BBBBBB - what I'm particularly trying to avoid is the scenario below where the screen is filled with mysterious black and white symbols, because unknown to the user, the default color #000000 is the same color as the black areas to which it's being compared, and black text, which is the common default. It's not intuitive what one is to do with this.

[image]

#BBBBBB is a better choice, and gives:

[image]

jimmiedave, 2022-02-28:
And of course, the highlighted swatch should be #BBBBBB, assuming it's in the  swatches.

neilweinstock, 2022-02-28:
> I'm fine with the #BBBBBB - what I'm particularly trying to avoid is the scenario below where the screen is filled with mysterious black and white symbols, because unknown to the user, the default [...]

To be fair... there are already about 20 other ways the color picker is not intuitive. ;)

But I agree, the black default is not ideal, the grey will be better.  And this will save me from changing every component manually to gray when I first enter a design, so I'm all for it. :)

SiboVG, 2022-02-28:
> And of course, the highlighted swatch should be #BBBBBB, assuming it's in the swatches.

#BBBBBB is not in the swatches, but either way if I set the default color to a color that is in the swatches, that swatch did not get selected, so let's just silently forget about it 😉

closed 2022-02-28 by SiboVG

neilweinstock, 2022-03-01:
The swatches don't highlight, and it's also happy to have the same color
appear multiple times in the "recently used" grid on the right side, and if
you choose a color in one of the other panels it does *not* show up in the
recently used grid.

It is not a good color picker, but it's what we seem to be stuck with for
now.

PR #1200 (2022-02-28): [fixes #1192] Use default 3D component color mid-grey + 30% shine
This PR fixes #1192 by setting the default 3D component color to a mid-grey #7F7F7F + default 30% shine.

Preview:
[image]
commit 746c251c (2022-02-28): [fixes #1192] Use default 3D component color mid-grey + 30% shine
commit dd864850 (2022-02-28): Merge pull request #1200 from SiboVG/issue-1192

[fixes #1192] Use default 3D component color mid-grey + 30% shine