openrocket/openrocket #1204: [Bug] Stage CG override not changing distance in design window, not changing stability

hcraigmiller, 2022-03-01:
Use of the Stage override is not changing the CG distance in the design window (upper-right corner), and the stability is not changing there either.

JoePfeiffer, 2022-03-05:
Sorry I didn't look at this one before:  near as I can tell, it is working as intended.  The CG override is for the empty stage; adding motors then changes it.  This way we don't need a different override for every different motor.

hcraigmiller, 2022-03-05:
After reading the comments of @JoePfeiffer, I believe that he is correct, it is functioning properly. I didn't realize at the time that the motor was causing the disparity. The override should not override the effect of the motor mass.

JoePfeiffer, 2022-03-05:
@hcraigmiller should this be closed?

PR #1223 (2022-03-06): [fixes #1204] Fix stage CG override incorrect location
This PR fixes #1204 by correcting for the CG location when it is overridden in a stage. The problem was that the CG overriding had no effect on the motor calculation, which this PR now adds.

https://user-images.githubusercontent.com/11031519/156888505-34976113-f784-462f-be64-fa6ae7c62252.mov
commit c98b90a8 (2022-03-06): Merge pull request #1223 from SiboVG/issue-1204

[fixes #1204] Fix stage CG override incorrect location
closed 2022-03-06 by JoePfeiffer