openrocket/openrocket #1207: [Bug] Tube Fin flight profile significantly differs from 15.03

hcraigmiller, 2022-03-02:
TRF user **Neutron95** reported a disparity in flight data for tube fin rockets between 15.03 and the beta release. Using the example tube fin rocket included with OR, I ran flight comparisons and confirmed the report

[image]
.
The flight profiles are significantly different; the same disparity exists with the current unstable.

neilweinstock, 2022-03-02:
My test was on Windows, for what it's worth.

SiboVG, 2022-03-02:
Same behavior noted on macOS (note: you should re-run the simulations on the unstable version, because the present simulation results are cached results)

neilweinstock, 2022-03-02:
Yes, I was looking at cached results, oops.  I confirm this issue as well.

SiboVG, 2022-03-07:
FYI: This issue is a result of commit [74ec0be340baf06d63cdc0e81cfd45e79bc2fa48](https://github.com/openrocket/openrocket/commit/74ec0be340baf06d63cdc0e81cfd45e79bc2fa48)

hcraigmiller, 2022-03-08:
To begin, 15.03 does not support tube fins. The user used a hack to simulate this feature, using launch lugs. The 15.03 results are at least 2x as high as both the unstable and Rocksim 10, using the user’s .ork file.

[image]
 
This does not look like an issue with the current unstable.

SiboVG, 2022-03-08:
Then I would suggest to update the simulation of the Tube Fins example design with unstable's results.

neilweinstock, 2022-03-08:
Wait, now I'm confused on multiple fronts.

1) Why do you say 15.03 didn't support tube fins?  It certainly did.  It was labeled "experimental" but it was there, using the same (I believe) approximation for simulating them as Rocksim (personally I would remove the "experimental" tag at this point but that's a minor detail).
2) I downloaded the ORK file he attached to his TRF posting, and it definitely used Tube fins.  Where is the ORK file that used a launch lug hack?
3) When this was first posted, we did tests on the posted file and verified the discrepancy.

neilweinstock, 2022-03-08:
Well, in this case the user said that 15.03 was already slightly *underestimating* the altitude on his tube fin rocket, and now we cut the estimate in half.  I have no explanation for the moment why Craig's Rocksim example shows estimates even lower than the beta.

I think I would like to start with a clear explanation of _why_ the beta changed.  I see you identified the commit that caused the change (and the description certainly seems to support that) but I don't have a clear idea what was the nature of the change.

neilweinstock, 2022-03-08:
> I just wanted to have another indication of which simulation is the best, other than a user report, that's why I asked Craig if he would want to simulate it in RockSim (since my guess was that we can trust them?)

Uh, maybe?  My gut feel is that the simulation aspects of Rocksim are pretty well-developed, but it is also true that Rocksim is tremendously buggy.  I *think* that we're all using the same tube fin approximation though, which is (if I recall correctly), three fins with equivalent profile for each tube fin.  So the results shouldn't be that different.

I am still trying to understand where the ORK file with launch lugs used as tube fins came from.

> As far as I could tell, the biggest change was that after the mentioned commit, the instancecount of a TubeFinSet was no longer the default '1' of a RocketComponent, but was now changed to the number of fin instances (so e.g. '6' in the Tube Fin example). Changing the instanceount back to 1 provided the same simulation results as in OR 15.03. But the instancecount-change is not necessarily the root cause, could be something deeper which I haven't found yet.

Ah, that's interesting.  I would hope that the *default* instancecount shouldn't really make a difference for the case where the user overrode the default to 3 or 6 or whatever, unless it was still simulating as if the value were still equal to the default.  Some quick tests ought to reveal the answer to that.

Certainly, something very significant changed from 15.03 to the beta, we should endeavor to understand it clearly before making any determinations on this issue.

Overall I still find this whole situation very perplexing.

hcraigmiller, 2022-03-08:
@neilweinstock is correct about 15.03 supporting tube fins; I only looked at the icon and didn't realize that the tube fins had a nearly indistinguishable icon from launch lugs. I am going to do some additional testing using a variety of tube sizes to see if that makes a difference.

neilweinstock, 2022-03-08:
And you're right that in 15.03, the icons are *extremely* similar for the launch lug and tube fins.  In the beta the tube fin icons were updated, and now look much different from the lug.

However, both icons are... well they could use a bit of work.  I'll give that a go when I get a chance; obviously not urgent.

neilweinstock, 2022-03-08:
Yikes

JoePfeiffer, 2022-03-08:
I say we all build some tube fin rockets and collect some data.

hcraigmiller, 2022-03-08:
Here are the comparisons.
[image]
[image]
[image]
[image]
[image]

The bigger the airframe and more powerful to motor, the closer you get to the Rocksim 10 results. The results for the BT-80 tests are _impossible_.

neilweinstock, 2022-03-08:
Note that you have 0 delays for all the motors in Rocksim... that's gonna lose you a lot of estimated altitude.

neilweinstock, 2022-03-08:
Something is screwy with the Rocksim results in that last batch... longer delays resulted in *lower* apogee???

hcraigmiller, 2022-03-08:
Here's a BT-20 for comparison... Apogee says that a 0 delay on motors that are not plugged is the equivalent of no ejection charge.

[image]

neilweinstock, 2022-03-08:
Ah, OK.  Look at all we're learning today! :)

JoePfeiffer, 2022-04-02:
Real data!  I was only able to get data with B6 and C6 motors; I didn't have any A8-5's to get a long enough delay to reach apogee.  Collected with a Jolly Logic Altimeter 3, dead calm air, approximately 70F, launch rod at a 7 degree angle from vertical.

[tubular.zip](https://github.com/openrocket/openrocket/files/8403482/tubular.zip)
[image]
[image]
[FlightLog-edited.xlsx](https://github.com/openrocket/openrocket/files/8403488/FlightLog-edited.xlsx)

SiboVG, 2022-04-03:
Awesome!! Good job @JoePfeiffer!

JoePfeiffer, 2022-04-13:
Some more data from TRF thread https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-2-is-now-available.171295/post-2265497

PR #1312 (2022-04-27): Rewrite tube fin and launch lug drag calculations
Directly compute drag of tube fins and launch lugs using characteristics of air in tubes, rather than the "three fin approximation" for tube fins and an interpolation from a wire loop to a closed tube for launch lugs.

I am pleasantly surprised by the accuracy of the simulation for the one rocket I have data to compare against.  I expect as we get data from more rockets with tube fins this will need to be refined.  These are the results for the tube fin rocket described in issue #1207.
<table>
<tr><th>Motor</th><th>Flight 1</th><th>Flight 2</th [... cut]
closed 2022-04-27 by SiboVG