openrocket/openrocket #1210: Pod simulation wildly out of range

sjh114961, 2022-03-02:


sjh114961, 2022-03-02:
[Pod-Boosters.zip](https://github.com/openrocket/openrocket/files/8171762/Pod-Boosters.zip)
Modified existing model to have pods. Using a "c6 " motor configuration results in wild values simulation, see screenshots
[image]
[image]
.

SiboVG, 2022-03-02:
Confirmed behavior, even just re-running one simulation multiple times yields widely varying results

JoePfeiffer, 2022-03-03:
This was originally reported as a bug against parallel stages; it turns out it also affects axial stages and has been lurking.
[premature-separation.zip](https://github.com/openrocket/openrocket/files/8174785/premature-separation.zip)

hcraigmiller, 2022-03-03:
As I'm looking at the .ork, it is set for stage separation on lower stage ignition. That means the separation event is on the pad; that's not a premature ignition, _that's an event that should never be allowed_. Or, am I missing something? 

If you set separation on lower stage ignition, plus at least 1 second, the simulation runs correctly; otherwise it does not.

sjh114961, 2022-03-03:
Since this ork is only parallel staging, setting the separation to be "Never" seems to work.

JoePfeiffer, 2022-03-03:
> As I'm looking at the .ork, and it is set for stage separation on lower stage ignition. That means the separation event is on the pad; that's not a premature ignition, _that's an event that should never be allowed_. Or, am I missing something?
> 
> If you set separation on lower stage ignition, plus at least 1 second, the simulation runs correctly; otherwise it does not.

I'm guessing the purpose is to allow exactly what you did -- set separation for some fixed time after liftoff.  Regardless of whether it should be allowed, it shouldn't be the default for parallel staging and it *certainly* shouldn't break the simulation (rockets can break.  The simulation can't!).

Haven't looked into the details yet, but I think terminating the simulation branch if a stage starts tumbling under boost would be a reasonable fix for the instability.

PR #1221 (2022-03-07): Correctly identify when time is past end of thrustcurve so thrust is 0 in getAverageThrust
What looked like numerical instability in simulations with side boosters turned out to be two bugs involving thrust calculations:

1. When calculating motor thrust, the end of the thrustcurve was not being correctly identified, so the motor thrust was effectively random for times after that..

2. If sustainer and side boosters were firing after stage separation (requires a bad configuration, but we should still do the right thing with bad configurations), thrust for all motors was used for calculation of [... cut]
closed 2022-03-07 by SiboVG