openrocket/openrocket #1211: Booster example is just plain bad

sjh114961, 2022-03-02:
1) I132 motor does not exist in motor selection
2) Core stage is 1.89in or 47.22mm. As far as I am aware there are no motors that size.

These are the example users are using to learn this new version. They HAVE to be right!

hcraigmiller, 2022-03-02:
Please provide an .ork file for what you believe would be an appropriate beginner booster example.

sjh114961, 2022-03-02:
I would be glad to, however, booster simulations are either failing or causing exception. In the case below an exception. So it would be better to wait until that issue is resolved.

JoePfeiffer, 2022-03-02:
I am able to select an Aerotech I132 in the motor selection. But other than that, you're right -- yes, the motor is going to rattle around in that body tube, and the design is treating the payload fairing as an unpowered "stage".

sjh114961, 2022-03-28:
The reason I did not see the I132 is that I had the hide OOP motor checked, my bad. I have attached the upgraded model. This model does appear to simulate correctly. Please review
[Booster Example.zip](https://github.com/openrocket/openrocket/files/8366461/Booster.Example.zip)
.

JoePfeiffer, 2022-05-07:
Looking at both the original and your improved version, I think what's really needed is a completely new example rocket that is only demonstrating parallel boosters.  This example is too complex, and doesn't really demonstrate parallel boosters since they never separate.  @sjh114961 , would you like to take a crack at a totally new, better example?

sjh114961, 2022-05-09:
Been in hospital, that's why I haven't responded. So it does have to separate. I will take a crack at it

sjh114961, 2022-05-15:
[Booster Example.zip](https://github.com/openrocket/openrocket/files/8695887/Booster.Example.zip)
Here is my updated example. Feel free to modify it as required, just let me know what you did.