openrocket/openrocket #1212: Booster simulation causes exception

sjh114961, 2022-03-02:
[Pod_Example.zip](https://github.com/openrocket/openrocket/files/8172668/Pod_Example.zip)
First run simulation failed to run.
[image]
[image]

Copies motor configuration
[image]

Went to simulation and got exception 
[image]

Bug report to follow 

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120) at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:414) at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:98) at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847) at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857) at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317) at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233) at java.desktop/java.awt.Dialog.show(Dialog.java:1070) at java.desktop/java.awt.Component.show(Component.java:1716) at java.desktop/java.awt.Component.setVisible(Component.java:1663) at java.desktop/java.awt.Window.setVisible(Window.java:1031) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005) at net.sf.openrocket.gui.main.SimulationPanel.activating(SimulationPanel.java:656) at net.sf.openrocket.gui.main.BasicFrame.stateChanged(BasicFrame.java:1734) at net.sf.openrocket.gui.main.BasicFrame_changeAdapter.stateChanged(BasicFrame.java:1747) at java.desktop/javax.swing.JTabbedPane.fireStateChanged(JTabbedPane.java:418) at java.desktop/javax.swing.JTabbedPane$ModelListener.stateChanged(JTabbedPane.java:269) at java.desktop/javax.swing.DefaultSingleSelectionModel.fireStateChanged(DefaultSingleSelectionModel.java:143) at java.desktop/javax.swing.DefaultSingleSelectionModel.setSelectedIndex(DefaultSingleSelectionModel.java:74) at java.desktop/javax.swing.JTabbedPane.setSelectedIndexImpl(JTabbedPane.java:622) at java.desktop/javax.swing.JTabbedPane.setSelectedIndex(JTabbedPane.java:597) at java.desktop/javax.swing.plaf.basic.BasicTabbedPaneUI$Handler.mousePressed(BasicTabbedPaneUI.java:4090) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6632) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4544) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.lang.NullPointerException at net.sf.openrocket.aerodynamics.BarrowmanCalculator.getDampingMultiplier(BarrowmanCalculator.java:829) at net.sf.openrocket.aerodynamics.BarrowmanCalculator.calculateDampingMoments(BarrowmanCalculator.java:780) at net.sf.openrocket.aerodynamics.BarrowmanCalculator.getAerodynamicForces(BarrowmanCalculator.java:189) at net.sf.openrocket.simulation.RK4SimulationStepper.calculateForces(RK4SimulationStepper.java:448) at net.sf.openrocket.simulation.RK4SimulationStepper.calculateAcceleration(RK4SimulationStepper.java:327) at net.sf.openrocket.simulation.RK4SimulationStepper.computeParameters(RK4SimulationStepper.java:293) at net.sf.openrocket.simulation.RK4SimulationStepper.step(RK4SimulationStepper.java:124) at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulateLoop(BasicEventSimulationEngine.java:138) at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulate(BasicEventSimulationEngine.java:90) at net.sf.openrocket.document.Simulation.simulate(Simulation.java:373) at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:55) at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:24) at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304) at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264) at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343) at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) at java.base/java.lang.Thread.run(Thread.java:829)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.01
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
[... cut ...] va.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) > at java.base/java.lang.Thread.run(Thread.java:829)
34613 1206.092 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog
34614 1206.092 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
34615 1262.744 USER (SwingExceptionHandler.java:220) User requested sending bug report
34616 1262.745 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2022-03-02:
Duplicate of #1184

closed 2022-03-02 by SiboVG

JoePfeiffer, 2022-03-02:
Picked up the design and attached it to #1184