openrocket/openrocket #1217: International Standard Atmosphere checkbox doesn't set ISA

JoePfeiffer, 2022-03-03:
In the Simulation Options Launch tab, there is an option to use the International Standard Atmosphere, which ought to set the temperature and pressure for the launch site according to that model. Instead, it locks the temperature and pressure however they were set before checking the box.  Pressure does change with altitude (see #982), but the ground level pressure isn't correct.

SiboVG, 2022-03-14:
Could you clarify the issue? If in 'A simple model rocket', I set the temperature to e.g. -19 °C and pressure to 950 mbar, with the ISA checkbox unchecked, the simulation indeed starts at the mentioned temp and pressure. If I then check the ISA checkbox, the simulation's temp and pressure start at 15 °C and 1013 mbar. What am I missing here?

JoePfeiffer, 2022-03-15:
Hmm.... you're right.  It's a flaw in the interface to the simulation editor, not an actual error.  When I set an arbitrary temperature and pressure and then click "Use International Standard Atmosphere", the values I set in the edit dialog remain unchanged there, even though after running a simulation I can check and the ISA was indeed used. So the boxes I've circled in red are misleading.
[image]

SiboVG, 2022-03-15:
Maybe misleading, but not per se a behavior I'd like to change. I find it useful to still have my user-entered values. Also want to note that the disable spinner box is more greyed out on macOS, so it's more clear than on Linux that those values aren't used.

closed 2022-03-15 by SiboVG