openrocket/openrocket #1220: Rocksim import miniaturizes fins

neilweinstock, 2022-03-04:
Reported here: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-is-underway-download-it-now.171295/post-2249624
[8 in Goblin 4 x 75mm.zip](https://github.com/openrocket/openrocket/files/8188751/8.in.Goblin.4.x.75mm.zip)

In Rocksim:
[image]

After import to OR:
[image]

hcraigmiller, 2022-03-04:
I'm not sure this is an OpenRocket problem. If I strip the AGM down to the outer frame (and removing fin tabs) in Rocksim 10, then save it to a new file, and import that into OpenRocket, then the fins shrink.

If I take the stripped AGM in Rocksim 10, delete the back fins, and recreate identical back fins, then import into OpenRocket, only the front fins shrink.

If I take that in Rocksim 10, delete the front fins, and recreate identical front fins too, then import into OpenRocket, none of the fins shrink shrink.

[image]

This may be an issue within Rocksim versions. But, still playing with it.

neilweinstock, 2022-03-05:
It seems to me that if, in any circumstance, the same RKT file looks different after import into OR than it does in Rocksim, then it's an OR problem.

I note that not only are the Goblin fins miniaturized, they are also the wrong shape.  In the Rocksim file there's a point list that looks like this:

`<PointList>329,0|379,277|151,370|0,0|0,0|0,0|0,0|0,0|0,0|0,0|</PointList>`

That definitely describes those huge Goblin-shaped fins (measurements in mm), but the OR importer is somehow mangling them into tiny parallelograms.

hcraigmiller, 2022-03-05:
I would disagree somewhat. If an old Rocksim file is not currently formatted correctly (because of no longer used code or otherwise), but when created in the current Rocksim version it works correctly, then that is a Rocksim problem, not an OpenRocket problem.

That said, I do agree that identification of the problem (comparing the data pre and post fin recreation) is needed to validate the conclusion (or improve OpenRocket conversion if appropriate). I am undertaking that now.

neilweinstock, 2022-03-05:
If Rocksim can read it, then we should be able to read it. Obviously there are practical limits to this, but in this case the correct data is there in the file, plain as day.

hcraigmiller, 2022-03-05:
Okay, this is not an OpenRocket problem, it is a Rocksim problem. Rocksim is intermittently adding additional points to the points list. Rocksim sometimes saves the file with too many |0,0| point sets, to such an extent that it is incompatible with OpenRocket. 

For example: <PointList>329,0|379,277|151,370|0,0|0,0|0,0|0,0|0,0|0,0|0,0|</PointList> should be <PointList>329,0|379,277|151,370|0,0|</PointList>
 
Everything after the first |0,0| point should be deleted. Although OpenRocket can handle two |0,0| points, instead of one, when the third is added OpenRocket shrinks the fins.

If the OpenRocket import conversion truncates anything after the first {0,0} point, it should work reliably.

PR #1224 (2022-03-06): [fixes #1220] Fix RockSim fin import issue
This PR fixes #1220, which was a RockSim fin import issue where the fins would get shrunk down. Thanks to @hcraigmiller for the [analysis of the problem](https://github.com/openrocket/openrocket/issues/1220#issuecomment-1059686384).

New import result:
[image]

Here is a [jar file](https://github.com/openrocket/openrocket/suites/5546831963/artifacts/178701675) for testing.
commit 933d2110 (2022-03-06): Merge pull request #1224 from SiboVG/issue-1220

[fixes #1220] Fix RockSim fin import issue
closed 2022-03-06 by JoePfeiffer